cyoung/stratux #440: Error displaying callsign (9b1 and some earlier)

rapgood, 2016-06-13:
1. Stratux version: 9b1
2. Stratux config: from img

    SDR
      [ ] single
      [x] dual

    GPS
      [x] yes
      [ ] no
      type: RY835AI

    AHRS
      [x] yes (with known limitations)
      [ ] no

    power source: wall

    usb cable: WiFi

3. EFB app and version: (e.g., WingX Pro7 8.6.2)
Foreflight v7.6.1
    EFB platform: (e.g., iOS 9.2)
iOS 9.3.2
    EFB hardware: (e.g., iPad Mini 2)
iPad 4
4. Description of your issue:
Periodically, Stratux will improperly format the callsign of a flight. E.g., I just saw a flight displayed as "HH0495" on Stratux that FlightAware properly displayed as "CHH495." Depending on the flight (I cannot discern the precise scenario), this improper formatting appears to happen only when "Show Traffic Source in Callsign" is TRUE.  I have recreated this by toggling "Show Traffic Source in Callsign" on other flights.

I attempted to enable logging by setting "Record Play Logs" to TRUE, but no log files appeared to be created in the expected directory "http://192.168.10.1/logs/stratux/"

cyoung, 2016-06-13:
Thanks for upgrading to v0.9b1 with the image.

Did you have logging turned on? If so, could you post/send http://192.168.10.1/logs/stratux.sqlite, please? If not, could you enable logs, reproduce it, and send the logs with a flight to check.

rapgood, 2016-06-13:
Will reproduce and send stratux.sqlite.

rapgood, 2016-06-13:
Reproduced error by:
1. Enabled "Record Replay Log"
2. Toggled "Show Traffic Source" between TRUE and FALSE several times
3. Disabled "Record Replay Log"

Performed "ls -al" on /var/log directory
stratux.sqlite shows zero bytes in length.

rapgood, 2016-06-13:
I, of course, waited for callsign that showed anomaly before steps 2 and 3. ;)

rapgood, 2016-06-14:
I also note that, sometimes, a callsign will be incomplete.  E,g., flight "AAY1805" shows in stratux as only "1805" (Show Traffic Source is FALSE) -- don't know if this is related problem or not.

rapgood, 2016-06-17:
Any theory as to why logging apparently won't work on my RP?

closed 2019-01-17 by Helno