cyoung/stratux #441: Is "time" available from EFB connection protocol?

bradanlane, 2016-06-17:
1. Stratux version: v0.9x

    SDR: single or dual
    GPS: no
    AHRS: n/a
    power source: n/a
3. EFB app and version: FF

There is this thread on Reddit ...
https://m.reddit.com/r/stratux/comments/4ogqzf/i_think_i_figured_out_why_i_always_got_erroneous/

It points to erroneous altitude in FF when Stratux does not have an up-to-date clock value. 

I know it's been discussed in the past. Without a GPS or internet connection, the Raspberry Pi can't update its clock. One option would be to have the WebUI automatically (and silently) check "Stratux time" to "mobile device time" and send back a useable time. This requires the user to manually start the WebUI. Still, it's worth doing under the covers. I'm happy to code it and submit a pull request.

Is there any chance the connection protocol of the EFB to Stratux contains a timestamp? If it is possible, this is the ideal solution as it is both automatic and transparent to the user.

ChiefPilot, 2017-04-07:
I ran into a similar thing as well when getting stratux to talk to a certified device.   For reasons unrelated, a separate GPS just for stratux wasn't possible so I took the approach of adding a flag to track whether or not the time had been set, then checking the flag when an HTTP request had been received.   If the flag wasn't set, set it using the Date http header and set the flag.

No end user change required, uses only standard HTTP headers, and best of all didn't require SSH'ing into the box each time to set the current date.   (The certified box needs accurate date/time else it marks data as "degraded").

Does this approach make sense or is it clownshoes?   I did it as a quick work-around while I pursue a more permanent solution to the issue, but perhaps it has broader application.

cyoung, 2017-04-19:
Could be, but we moved away from using any external clocks in the code (except for GPS attitude) so if the clock is off on the RPi, it should not affect anything.

closed 2017-04-19 by cyoung