cyoung/stratux #443: Is it time to promote v0.9b1 to be the stable release?

JohnOCFII, 2016-06-21:
The current listed beta version, v0.9b1 seems stable for me, and I don't recall seeing too many issues either here or on Reddit.  Should it be promoted to be the current stable version?

cyoung, 2016-06-21:
No real show stoppers and it has been 25 days, I agree.

Ergonomicmike, 2016-06-22:
I agree. I've got about 10 hours on the later .sh's with no problems. (I haven't been back to KPAN yet to do more testing with FIS-B buffering.)

cyoung, 2016-06-22:
You call it @Ergonomicmike - is it v1.0 material yet?

Ergonomicmike, 2016-06-23:
Oops, just saw your question to me here. 

Yikes - I'm flattered.  But since I don't know code,  I don't think that I should be the one making the "ship it" call for v1. But I might call it a v0.9r1 soon. (I haven't flown with the last few commits yet. Let's wait until after Saturday.)

Having said that, there's this from the [reddit](https://www.reddit.com/r/stratux/comments/4o8t62/unexplained_wifi_stoppages_in_flight/): "It appears the FIFO task to send UDP crashes for some reason. Both v0.8r2 and v0.9b1 do the same thing when the message exceed a limit. Crashes as no message are sent, yet the wifi is up and I can see the access point, however no connection can be established to reset anything, you must power cycle to recover the device to work again. Mine crashing with in a short period as the messages just stop. The Pi3 and still running. Avare I/O helper stops and will start receiving messages on powercycle and access point connection. Something is wrong with I/O buffering in the stratux software."

And too, now there's new issue https://github.com/cyoung/stratux/issues/447, which might be a limitation of the vk's (as opposed to Stratux). But, since the highest I've been in the Glasair is 15,500, someone else will have to gather data. There are other Lear guys here - are they having similar problems with Stratux GPS alt abv FL 330?

JohnOCFII, 2016-06-23:
> Yikes - I'm flattered. But since I don't know code, I don't think that I should be the one making the "ship it" call for v1. But I might call it a v0.9r1 soon. (I haven't flown with the last few commits yet. Let's wait until after Saturday.)

Just to be clear, my original note was suggesting that the current v0.9b1 be promoted to replace the current v0.8r2 (perhaps as v0.9r1).  I was not suggesting that the later commits have been tested enough to be called a release version.  So if @cyoung is suggesting that something be v1.0, I believe he discussing the current zip file for v0.9b1, not anything more recent.

Ergonomicmike, 2016-06-23:
@cyoung After thinking about it more (and reading more reports in the github & the reddit), I agree with John and think it would be good to make a later .sh version of v0.9b1 be r1. (Like, before the weekend if possible?) The driving factor is the AIRMET hang with old v0.8r1. 

Having siad that, I don't know which snapshot to make r1. I flew Tuesday morning with 1c7a, and it worked fine for me. (Benign weather.) But it looks like a few of the later commits are important for UAT problems. But, not having flown the latest, I can't call it. So perhaps promote 1c7a to v0.9r1 and let users bang on it over the weekend?

In the meantime, it would be cool to have a v1.0 in time for OSH. But I went down that road with the Glasair. It's not done until it's done.

cyoung, 2016-06-23:
The gen_gdl90 process restart addition made recently should be included in the next stable release version. I'll make a beta release shortly with what we have currently.

ScraboTower, 2016-06-25:
I see more "GPS Solution : Disconnected" messages on 9b1 than on 8r2 and WingX "Status" lights turn red for a second or two for both location and altitude.

FYI : GPS & AHRS connected via GPIO.

Ergonomicmike, 2016-06-26:
I flew 2 hours today w/ v0.9b2. I didn't look at the web UI more than a few times for a few seconds, but operationally, as an appliance, the Stratux worked fine for me. BU-353 & iFly.

Ergonomicmike, 2016-06-26:
@ScraboTower Is the Air Force still degrading GPS in our area?

ScraboTower, 2016-06-26:
Not today at least according to this. https://www.faasafety.gov/files/notices/2016/Jun/CHLK_16-08_GPS_Flight_Advisory.pdf

Ergonomicmike, 2016-06-26:
So did you have logging enabled, a la Axtel4 in #449?

JohnOCFII, 2016-07-03:
I'm still thinking the community would benefit by replacing v0.8r2 with v0.9b2.  The folks I help build are naturally downloading the "r" Release versions and not the "b" Beta versions.  I've been upgrading the software for them.

AvSquirrel, 2016-07-15:
@JohnOCFII - commit 945c8bf6 was elevated to v1.0r1 on Wednesday afternoon. Can we call this closed?

JohnOCFII, 2016-07-15:
The Reddit page needs to be updated too.  Still people loading up v0.8r2, reporting issues, and having to upgrade.

https://www.reddit.com/r/stratux/

cyoung, 2016-07-15:
Fixed, thanks

closed 2016-07-15 by cyoung