cyoung/stratux #445: Automatically respawn gen_gdl90 on failure

ssokol, 2016-06-21:
1. Stratux version: v0.8r2 / v0.9r1
2. Stratux config: Any

    SDR
      [X] single
      [X] dual

    GPS
      [X] yes
      [X] no
      type: N/A

    AHRS
      [X] yes
      [X] no

    power source: any

    usb cable: any

3. EFB app and version: any

    EFB platform: N/A (any)

    EFB hardware: N/A (any)

4. Description of your issue: Suggestion / Enhancement

If and when the gen_gdl90 process dies it should automatically restart (at least on "production" rather than "development" systems). The recent failure caused by the change in the AIRMET data made this abundantly clear. Lots of people spent lots of time restarting their Stratux system. We need to revise the initialization system to include either an automatic respawn directive or have gen_gdl90 spawn a watcher daemon - whichever is easier.

The current version of the Stratux init script is legacy, dating from the Pi 2 / Wheezy days, and lacks any kind of respawn functionality. Jessie is based on systemd and making a daemon automatically respawn in a single value in the declarative structure of the startup script. BUT -- much of the update code / logic is part of the legacy stratux init script. That being the case, it may be easier to stick with the legacy startup and have it spawn a watcher.

Thoughts?

kdknigga, 2016-06-21:
I've actually implemented this with no problems.  I went the native systemd
unit file route.  It wasn't hard at all.  And I successfully demonstrated
it with a real in-flight gen_gdl90 crash last Friday.

kdknigga, 2016-06-22:
I think this should be enough, but I haven't tested the update script.

https://github.com/cyoung/stratux/compare/master...kdknigga:master

commit 5b998896 (2016-06-22): Merge pull request #446 from kdknigga/master

convert from sysv init to systemd for the crash recovery benefits. Fixes #445.
closed 2016-06-22 by cyoung