cyoung/stratux #447: Altitude issue with usb dongles.

flyinslow, 2016-06-23:
1. Stratux version:9b1
2. Stratux config:IMG

    SDR
      [ ] single
      [x ] dual

    GPS
      [ x] yes
      [ ] no
      type:Vk 162 and vk172

    AHRS
      [ ] yes
      [x ] no

    power source:wall

    usb cable:wifi

3. EFB app and version: foreflight v7.6.1

    EFB platform: iOS 9.3.1

    EFB hardware: iPad Air 2

4. Description of your issue:I am getting the altitude discrepancy when passing through about 32780. The issue is occurring while using the gps dongle. If I use the iPad gps, the altitude is accurate. For those asking why I don't just use the iPad gps, it's reception is spotty at best, making it useless most of the time. Everything else in 9b1 seems to be working fine. 

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

AvSquirrel, 2016-06-23:
Known bug, caused by an overflow on the geometric altitude message. Closed on our end; similar bug is open on the current (v7.6.1) public version of ForeFlight.

On the Stratux side that caused geometric height to be converted to int16 before encoding into the 16-bit format used in the GDL90 message. This caused altitude to roll over above 32767'. On the Stratux side, it was identified with #373, fixed at #375, and has been in the release images since v0.9b1.

By coincidence, ForeFlight v7.6.1 also has a similar bug on its decoding of these messages. This is known to the ForeFlight team, but no new public releases have been made since it was reported. 

@bsdcat -- can you confirm if a fix has been committed in your beta builds?

Ergonomicmike, 2016-06-23:
Thx. That leaves a possible FIFO problem for UAT messages that I mentioned in #443. (Or perhaps the users that are having disconnects are having sleepflag issues?)

bsdcat, 2016-06-23:
Yes, it has.

bsdcat, 2016-07-14:
To follow up, we released ForeFlight Mobile 7.7 and it handles altitudes over 32,767' in the ownship geometric altitude message correctly.

daniellwu, 2016-10-10:
so can this ticket be closed per Foreflight update?

cyoung, 2016-10-10:
@daniellwu - Yes, thanks!

closed 2016-10-10 by cyoung