cyoung/stratux #448: What is the proper procedure for updateing the .sh files via wireless?

Axtel4, 2016-06-24:
1. Stratux version:  v0.9b1-3250158126.sh
2. Stratux config:

    SDR
      [ ] single
      [ X] dual

    GPS
      [ X] yes
      [ ] no
      type:

    AHRS
      [ ] yes
      [ X] no

    power source:NA

    usb cable: NA

3. EFB app and version: (e.g., WingX Pro7 8.6.2) NA

    EFB platform: (e.g., iOS 9.2) NA

    EFB hardware: (e.g., iPad Mini 2) NA

4. Description of your issue:
Today, I tried updating my Stratux via wireless for the first time.  I downloaded the current .sh file and tried to update through the Web UI tool.  Stratux went off line (the 60 second message) and never connected the Web UI again.  I was able to connected to the Stratux AP but 192.168.10.1 was never available.  I had to pulled the SD card, reimaged, and updated via wired to recover.  I was hoping to just walk out the hangar and update the Stratux in the aircraft remotely via WiFi, but it looks like I be carrying them home to update via a wired connection.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

cyoung, 2016-06-24:
I haven't timed it but maybe 60 seconds isn't enough. It reboots twice. How much time did you give it?

Ergonomicmike, 2016-06-24:
I have found that, depending on the browser, you sometimes have to clear the browser's cache before the web UI will reinitialize. Also, as Chris said, I've learned thru experience to wait for the 2nd appearance of the Wi-Fi signal. (i.e., the 2nd reboot.)

Axtel4, 2016-06-24:
@young: About 15 minutes.  I was busy doing other things and came back to it later to find it would access the Web UI page.

@ErgonomicMike: The first step was to clear the browser cache and try it, no joy.  Moved it over to the wired connection, rebooted Stratux, and still could not access the Web UI through the browser.  I was able to SSH into the R-Pi though.

cyoung, 2016-06-24:
Which version were you upgrading from?

Do you know image version (not including .sh updates) that it was running before?

We're using systemd to start stratux now, so pre-Jessie images won't take the .sh files.

Axtel4, 2016-06-24:
This one: v0.9b1-3250158126.sh.  I flew 4 hours with it yesterday and you push a new .sh file when I got back home.

I was running the 0.8r1 image updated to v0.8r2-7dca721b4c through the sh update process, then finally updating to v0.9b1-3250158126.sh on Wednesday evening.

cyoung, 2016-06-24:
It's because your image was running Wheezy from pre-v0.8r2. In v0.8r2 image we upgraded the operating system to Jessie. We made a switch that breaks things if you're using the old operating system. I'm undecided on if we want to fix it and let the .sh files update the old versions forever or let it break. There are some updates that would be impractical to include in the .sh file.

Axtel4, 2016-06-24:
So, I should download the 9.2 image and start from there then?

If that is the case, then you might want to let others know so they don't fall into the same trap when updating from an older .sh'd up image.

cyoung, 2016-06-24:
Yes, or even starting from v0.8r2 which is 3 months old now. I'll change the .sh file so it at least doesn't break the process and add something in the UI prompting the user to re-image.

skypuppy, 2016-06-24:
There are many bug fixes and performance enhancements between Wheezy and 
Jessie, with Jessie being the newest, of course.

Axtel4, 2016-06-24:
Since I need to reimage, I'll jump to 0.9b2 and update from there.

Are there any other changes I need to make to my Git / Make / Install (Development) Stratux besides updating to the 0.9b2 image?

cyoung, 2016-06-24:
Nope, the image has everything you need.

cyoung, 2016-06-24:
[image]

cyoung, 2016-06-24:
Making this v0.9b3.

Axtel4, 2016-06-24:
Thanks.  I have DL 0.9b2 and will sh up to 0.9b3.  I'll take it flying later after the TSRA passes by our area.   I plan on flying out for some BBQ.

Ergonomicmike, 2016-06-25:
TFR? (You said TRSA.)

kdknigga, 2016-06-25:
TSRA

You don't have thunderstorms in Arizona?

Ergonomicmike, 2016-06-25:
Oh, METAR TSRA's. I thought you were waiting for Biden to leave.

Axtel4, 2016-06-25:
I'm in flyover country. None of them hardly ever come around here. We might be full election cycles between TFRs.

Axtel4, 2016-06-25:
@cyoung: You can close this if you want. I have 0.9r3 loaded.  I'll need to wait for the next sh release to try it wireless again.

cyoung, 2016-06-25:
I think the fact that it was over wireless was just a coincidence, there definitely was an issue in the update via wireless or wired that would have presented the same symptoms.

Thanks for the report.

closed 2016-06-25 by cyoung

Axtel4, 2016-06-28:
I downloaded v0.9b3-31464f92ad.sh and attempted a wireless update. The reported version is sill at f4a36b after Stratux came back on-line.  I even cleared the Browser cache and reloaded, but it is still stuck on f4a36b.  The next step is to update via wired.