cyoung/stratux #449: 0.9b3 Trip report

Axtel4, 2016-06-26:
1. Stratux version:  0.9b3 stratux-v0.9b3-9d3060af95.sh
2. Stratux config: 

    SDR
      [ ] single
      [ X ] dual

    GPS
      [ X ] yes
      [ ] no
      type: VK-172

    AHRS
      [ ] yes
      [ N ] no

    power source:  Ship power

    usb cable:  Anker HD

3. EFB app and version: (e.g., WingX Pro7 8.6.2)   DroidEFB 2.1.3

    EFB platform: (e.g., iOS 9.2)  Android KitKat

    EFB hardware: (e.g., iPad Mini 2)  ASUS TF300 Table / LG G3 

4. Description of your issue:
I got about 4 hours of flying in with 0.9b3 today.  Prior to departure I started Stratux, connected the EFB devices and enabled logging (not verbose).

While on climb out to altitude (IFR flight plan) I received an EFB warning " No GPS signal or signal suspect" warning.  I switched to my device's internal GPS and waited until time would to troubleshoot.

Once at altitude, I switch to the Statux Web UI and saw a list of "WARNING SQLite lite logging behind".  I disabled logging and tried to connect to the GPS again.  No joy.  Note, at this time I was still receiving traffic, Weather and NexRad updates.  I had to turn off logging and report Stratux to recover the GPS connection. 

Turning logging back on it only took about 5 minutes for the problem to reappear.  I rebooted again and left logging of for the rest of my flight.  I did not have and GPS connection issues with the logging turned off.  Note: This happened with both devices I had connected to Stratux.

I'll come back later and add screen shots.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.  I'll see if anything was captured in the logs.

Ergonomicmike, 2016-06-26:
Kinda cross posting here, 'cause I said this under another issue. I flew 2 hours today w/ v0.9b2 (fresh image) with no issues. BU-353. Logging disabled. NEXRAD coming in. Guess I'll have to put "Enable logging" on my test card for next time.

Interestingly, no AIRMETs issued today, despite there being mod turb.

Axtel4, 2016-06-26:
No issues after turning off logging in 0.9b3.  The my entire 2 hour return flight - sans logging - was without error.

Ergonomicmike, 2017-06-24:
Am thinking this has been fixed and so this issue can be closed? I inadvertently enabled sqlite logging a few days ago with v1.3.b2. Ran for at least an hour with no error messages.

cyoung, 2018-01-17:
The logging options were moved to "developer mode" only.

https://github.com/cyoung/stratux/commit/da28af53c08a8c0c6a4d004c3b3b84a89b11c3b9

closed 2018-01-17 by cyoung