cyoung/stratux #450: sh updates build error (already fixed?)

skypuppy, 2016-06-27:
1. Stratux version: 0.9b3
2. Stratux config:

    SDR
      [ ] single
      [ ] dual

    GPS
      [ ] yes
      [ ] no
      type:

    AHRS
      [ ] yes
      [ ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:  Files not found during .sh updates.  File for thokk attached.
[stratuxBuildErrors.gz](https://github.com/cyoung/stratux/files/335790/stratuxBuildErrors.gz)

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

jpoirier, 2016-06-27:
@skypuppy  uhm, this should be over at https://github.com/jpoirier/stratux-setup/issues

skypuppy, 2016-06-27:
Thanks.  I had no clue.

jpoirier, 2016-06-27:
I see" unrecognized import path ... (import path does not begin with hostname)" which usually means go has a problem with GOROOT. Did you install another golang instance using apt-get install golang?

If so then uninstall (sudo apt-get purge golang*) and rerun the script.

skypuppy, 2016-06-27:
Nope.  I don't remember installing it by hand but just in case, I ran the "purge" with zero results.  So no, no golang previously installed.

jpoirier, 2016-06-27:
Okay, thanks for checking. I'll do a test with a fresh image in about an hour.

jpoirier, 2016-06-27:
Strange, the go tools archive download was good but there was a problem with the unarchive process. Possible SD card problem?   

    2016-06-27 10:24:39 (3.83 MB/s) - ‘go1.7beta1.linux-armv6l.tar.gz.1’ saved [67842788/67842788]
    go/misc/tour/static/lib/jquery-ui.min.js
    go/misc/tour/static/lib/jquery.min.js
    ...
    gzip: stdin: unexpected end of file
    go/misc/tour/static/partials/
    go/misc/tour/static/partials/editor.html
    tar: Unexpected EOF in archive
    tar: Unexpected EOF in archive
    tar: Error is not recoverable: exiting now

skypuppy, 2016-06-27:
Your SD card or mine?  :)
Sometimes I put my RPi work on a USB hard drive (with it's own power supply) to lessen the # of writes/reads to the SD card.

jpoirier, 2016-06-27:
> Your SD card or mine? :)

It's your log file I'm looking through so I'm guessing it's your SD card.

Axtel4, 2016-06-28:
I added this to my closed "the proper way" thread.  Maybe it should go here. . .

> I downloaded v0.9b3-31464f92ad.sh and attempted a wireless update. The reported version is sill at f4a36b after Stratux came back on-line. I even cleared the Browser cache and reloaded, but it is still stuck on f4a36b. The next step is to update via wired.

> Edit: Tried to sh update through a wired connection. The 0.9r3 built persists at f4a36b. (I tried rolling up to v0.9b3-31464f92ad.sh and down to v0.9b2-d50748c23a.sh with the same results.) If I get some time this afternoon / evening I'll reimage and try again.

What logs would you need to view for this issue?

jpoirier, 2016-07-08:
closing; I'm unable to replicate the problem on either rpi-2 or rpi-3

closed 2016-07-08 by jpoirier