cyoung/stratux #451: Jessie and VK-172?

Axtel4, 2016-06-30:
1. Stratux version: v0.9b3-31464f92ad.sh (actually v0.8r2 and above)

2. Stratux config:

SDR

[ ] single

[ X] dual

GPS

[ X] yes

[ ] no

Type: VK-172

AHRS

[ ] yes

[ X] no

power source: Ship Power via MyGoFlight USB power adaptor

usb cable: Anker Heavy duty (1 ft bench testing, 3 ft aircraft)

3.EFB app and version: (e.g., WingX Pro7 8.6.2) NA

EFB platform: (e.g., iOS 9.2) NA

EFB hardware: (e.g., iPad Mini 2) NA

4.Description of your issue: What is unique about Jessie that does not let it play nicely with the VK-172 at start-up?  After updating to v 0.8r2 and progressing through 0.9b3 I have noticed the VK-172 takes forever to start after an extended period of being powered down (i.e > 3 hours).    After a series of tests, the VK-172 will on average take approximately 43 minute to initialize with v0.8r2 and above. During initialization the Status page will show "0 in solution, 9+ seen; 10+ tracked."  Sometimes I can force a quicker initialization of the VK-172 by power cycling Stratux when running v0.8r2 or greater.  

Using the same hardware put swapping the SD card image with v0.8r1-b2eb382c1a.sh the VK-172 takes on average 140 seconds to initialize.

My Stratux test hardware is using a Raspberry Pi 2, the Edimax WiFi dongle, two shielded Nono 2s, and a “bottom” shielded VK-172 GPS dongle.  The VK-172 is mounted above the 978 MHz assigned dongle while the WiFi dongle is mounted above the 1090 MHz assigned dongle.  Both image loads are on SanDisk Ultra 8 GB SD cards.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

cyoung, 2016-06-30:
Would you be able to measure the power consumption of v0.8r2+v0.9b3 vs. v0.8r1, please?

Axtel4, 2016-06-30:
I'll need to rig up a USB current test cable.  I knew I should have got a USB volt / amp meter when I placed my last Amazon order.  Let me dig through my junk box an see what I can put together and I'll get the measurements.

heelix, 2016-06-30:
I've got my line tester.  I can do a quick test tonight.

cyoung, 2016-06-30:
The USB volt / amp meters tend to cause even more trouble, I just use a multimeter now.

Axtel4, 2016-06-30:
Do just want voltage measurement, or a voltage and current measurement?  I can get the voltage measurement in a few minutes.  For the current I'll need to rig up a cable.

Axtel4, 2016-06-30:
Here are the voltage measurements.  I'll need to rig something up for the current measurement if you want those also.

I included the GPS stats after approximately 2 minutes after power-up.  I didn't see the GPS for 0.9b2 "jump" into WAAS mode like 0.8r1 did.  I went back and did some more testing and 0.9b2 might go into WAAS mode after 20 + minutes, whereas 0.8r1 took a mater of minutes.  The Stratux was setting on the same spot on my work bench in the house for all testing.

Measurements taken after boot and WiFi active. Anker 1ft HD cable. Driving a 5V fan on the 3.3V bus.

MyGoFlight USB converter
Port 1:
V0.9b2
R-Pi = 4.83V
VK-172 = 4.75V
3D GPS, 8 in solution,12 seen,13 tracked

V0.8r1:
R-Pi = 4.85
VK-172 = 4.77
DGPS, 7,10,12

Port 2:
V0.9b2
R-Pi = 4.81V
VK-172 = 4.73V
3D GPS, 8,8,13

V0.8r1:
R-Pi = 4.84 V
VK-172 = 4.76 V
DGPS, 7,10,14

Anker Power Drive 2 USB converter
Port 1:
V0.9b2
R-Pi = 4.84 V
VK-172 = 4.76 V
3D,8,11,13

V0.8r1:
R-Pi = 4.85 V
VK-172 = 4.77 V
DGPS, 7,8,13

Port 2:
V0.9b2
R-Pi = 4.85 V
VK-172 = 4.76 V
3D, 7,10,13

V0.8r1
R-Pi = 4.85 V
VK-172 = 4.76 V
DGPS, 7,9,13

Axtel4, 2016-07-01:
Both images draw about 1.06 amps.  With both builds and using my lab supply I was able to drive my Stratux set-up as low as 3.89 V input to the Pi.  (That was the lower voltage limit of my lab supply.)  The Red LED was extinguished, both SDRs received traffic and the GPS remained locked and would go into WAAS mode, and the Stratux would communicate via WiFi with my lap top.

Axtel4, 2016-07-04:
I am going to close this issue.  I have not been able to correlate the performance of the VK-172 with the Jessie version.  I plan to do some more testing and I will post those results in Issue 203 but it will be sometime after the Fourth of July holiday.  

Here are my test observations to date:
The VK-172 lock issue appears to have more to do with the R-Pi build than with the VK-172.  I have two R-Pi 2 B's that I have been testing.  Both R-Pi’s are P.R.C. built units and came from MicroCenter with one being an Element 14 unit and the other was a “no name” R-Pi from the “Bargain Bin” that I don’t think they carry anymore. Unfortunately I don’t remember which R-Pi was which and the CCAs for both are marked the same.

Swapping hardware and Stratux OS between the R-Pi’s I noticed one is consistent with the VK-172 locking on in a matter of minutes, whereas the other P.R.C R-Pi will take 10’s of minutes to acquire a GPS lock.  

Utilizing a stiff lab supply* and performing low voltage dropout tests, one R-Pi will operate consistently down to 3.84 V input, measured at Transorb D5.  When operating this R-Pi at 3.84V, the Red LED is not illuminated, but the Stratux retains WiFi connection, the GPS acquires and retains lock, and both SDRs receive traffic.  The other R-Pi has trouble maintaining a WiFi connection and GPS lock at 4.5 V. I have a UK built R-Pi in my RY-835 / NooElec Mini build.  My next plan is to extract the UK R-Pi from the Mini build and perform the same testing with it to see if it performs any different.

*Stiff lab supply is defined as a supply that sags less than 100 mV when the current is stepped from no load to 1 amp at the rated voltage.

closed 2016-07-04 by Axtel4

skypuppy, 2016-07-04:
@Axtel4, I  ran into the (trivial) same problem with ID'ing RPi's and Beaglebone Blacks.  My solution is to use a felt-tip permanent marker and write on the top of the ethernet connector.  My particular nomenclature is "U1-xx" for both RPi and BBB (because their physical differences are so obvious.  In one page of the engineering notebook I jot down that unit's name and whatever relevant data I wish to keep about it.  Mostly the numbering sequence simply corresponds to their order of purchase.  
Trying to do the same with SD cards but they are more problematic as their contents can change so easily.  :)
With so many different types of SOC's at my house, I had to do  something.  :)