cyoung/stratux #453: Stratux data to iPad cuts out after a few minutes.  Uses NooElec Mini2+ with antenna suppied by NooElec.  Any ideas on why, antenna ??

JM9055, 2016-07-06:
1. Stratux version:
2. Stratux config:

    SDR
      [ ] single
      [ ] dual

    GPS
      [ ] yes
      [ ] no
      type:

    AHRS
      [ ] yes
      [ ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

cyoung, 2016-07-06:
Stratux version?

JM9055, 2016-07-06:
I assume I need to insert the micro chip in the adapter to find version?

JM9055, 2016-07-06:
Date is 3-16-2016 on version program.

cyoung, 2016-08-09:
Should be fixed with v1.0r1, update to the latest version linked on the website (http://stratux.me/). Thanks.

closed 2016-08-09 by cyoung