cyoung/stratux #462: New version 1.0r1 the SH did not work from 0.9r3 using the web interface.

dbarrington1, 2016-07-15:
1. Stratux version: 0.9r3
2. Stratux config: Pi3 

    SDR
      [ ] single
      [ X] dual

    GPS
      [ X] yes
      [ ] no
      type:

    AHRS
      [X ] yes
      [ ] no

    power source: 2 amp wall wart 

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2) Forflight and Wing X 

    EFB platform: (e.g., iOS 9.2) IOS 9.2 or whatever is newest not related to this was using Windows 7 with Chrome on my desktop to do the SH from the web UI

    EFB hardware: (e.g., iPad Mini 2) Ipad Mini 4 128G Wifi

4. Description of your issue: New version 1.0r1 the SH did not work from 0.9r3 using the web interface.  It says success and I wait at least 60 seconds and then reconnect and refresh and it still says 9.3 and still gives the Foreflight not playing well in the sandbox warning that is removed from 1.0  I will be making a new image just wanted you to be aware.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

cyoung, 2016-07-15:
Could you try the update and wait, say, 10 minutes (without removing the power source)?

There are some changes the the .sh update makes sometimes that takes a while, depending on the state of the system.

kdknigga, 2016-07-15:
For what it's worth, it worked for me upgrading from 0.9b2.

dbarrington1, 2016-07-15:
Sorry by the time I got it I had already tried like 3 times and then went ahead and downloaded the complete image and now I have that running.

closed 2016-07-15 by cyoung