cyoung/stratux #473: stratux-help command does not exist

jdwall, 2016-08-12:
Stratux Version: `v1.0r1 (945c8bf6bc)`

**Stratux Configuration:**

1) SDR

- [X] Single
- [ ] Dual

2) GPS

- [X] Yes
- [ ] No

Type:  `VK-172`

3) AHRS

- [ ] Yes
- [X] No

4) Power Source

- [X] USB

Description of your issue:

This is a simple issue, but, the welcome message when logging in via SSH references `stratux-help` which is a command that does not exist. 

```
Jacobs-MacBook-Pro:~ jacobwall$ ssh pi@192.168.10.1
pi@192.168.10.1's password:

 ad88888ba  888888888888  88888888ba          db    888888888888  88        88  8b        d8
d8"     "8b      88       88      "8b        d88b        88       88        88   Y8,    ,8P
Y8,              88       88      ,8P       d8'`8b       88       88        88    `8b  d8'
`Y8aaaaa,        88       88aaaaaa8P'      d8'  `8b      88       88        88      Y88P
  `"""""8b,      88       88""""88'       d8YaaaaY8b     88       88        88      d88b
        `8b      88       88    `8b      d8""""""""8b    88       88        88    ,8P  Y8,
Y8a     a8P      88       88     `8b    d8'        `8b   88       Y8a.    .a8P   d8'    `8b
 "Y88888P"       88       88      `8b  d8'          `8b  88        `"Y8888Y"'   8P        Y8

NOTE TO DEVELOPERS: Make sure that your system has an acceptable clock source, i.e., a GPS
with sufficient signal or enable ntpd (internet connection required).

Everything here comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.

Type 'stratux-help' for a few debugging commands.
pi@raspberrypi:~ $ stratux-help
-bash: stratux-help: command not found
```

When looking at the source, there are two references to `stratux-help`. 

1) [/stratux/image/motd](https://github.com/cyoung/stratux/blob/master/image/motd)
2) [/stratux/image/stxAliases.txt](https://github.com/cyoung/stratux/blob/master/image/stxAliases.txt)

jdwall, 2016-08-12:
If I am not mistaken, the below would be the fix to [stxAliases.txt](https://github.com/cyoung/stratux/blob/master/image/stxAliases.txt)

Lines 28 → end

```
alias stratux-help='function _stxhelp() {
echo ""
echo "###################################################"
echo ""
echo "These are commands avalible to you to debug Stratux"
echo ""
echo "###################################################"
echo ""
echo "stratux-off               Shutdown Stratux"
echo "sa                        List all services running"
echo "stxrestart                Restart Startux service only"
echo "stxstop                   Stop Stratux Service."
echo "stxstart                  Start Stratux Service."
echo "sdr0                      Test to see if sdr0 is in use by Stratux."
echo "sdr1                      Test to see if sdr1 is in use by Startux."
echo "sdrs                      Test to see if both SDRs are in use by Stratux."
echo "d90                       List my SDRs with serials"
echo "d900                      Run dump1090 on SDR0 in interactive mode."
echo "d901                      Run dump1090 on SDR0 in interactive mode."
echo "kal0                      Find Channels for calibrating PPM of SDR0."
echo "kal1                      Find Channels for calibrating PPM of SDR1."
echo "kalChan0                  Use the Chan from above to get ppm pf SDR0. Example: kalChan0 151"
echo "kalChan1                  Use the Chan from above to get ppm pf SDR1. Example: kalChan1 151"
echo "setSerial0                Set the PPM error to SDR0. Value from kalChan0. Example: setSerial0 -45"
echo "setSerial1                Set the PPM error to SDR1. Value from kalChan1. Example: setSerial1 23"
echo "hostapd_manager.sh        Sets the Change SSID, Channel, or Encryption for your Stratux"
echo "raspi-config              Open Raspberry Pi settings to expand filesystem"
}; _stxhelp'
```

cyoung, 2016-08-16:
Oh boy, I am most excited about the checkbox formatting on the issue report. Thanks for that.

The problem is that the aliases file is only applied to the root user via the bashrc we copy to `/root/.bashrc`.

I've clarified the message in `/etc/motd` - most of the commands cannot be run as the user "pi". Execute `sudo su -` and then everything should be working as expected.

closed 2016-08-16 by cyoung

jdwall, 2016-08-16:
I was wrong on my fix. :P Glad that you clarified it. I feel sort of stupid to not have tried `sudo su -` before. Oh well, at least it provides clarity down the road for anyone else. :)