cyoung/stratux #474: Upgraded from .91 to 1.0, TP-Link no longer works

Nokomis449, 2016-08-14:
1. Stratux version:1r
2. Stratux config:Pi2, long SDR's, VK-172, TP-Link WiFi dongle

    SDR
      [ ] single
      [X ] dual

    GPS
      [X ] yes
      [ ] no
      type:

    AHRS
      [ ] yes
      [X ] no

    power source: ship

    usb cable: fat

3. EFB app and version: (e.g., WingX Pro7 8.6.2)
iFly latest beta
    EFB platform: (e.g., iOS 9.2)
Android
    EFB hardware: (e.g., iPad Mini 2)
LG tablet, Samsung phone, iFly 740 dedicated EFB
4. Description of your issue:
Upgraded last night to latest Stratux from .9.1.  WiFi would not work.  Swapped out the TP-Link and inserted an EdiMAX and it worked fine.
I looked in the non-Edi-MAX hostapd.conf file and there is no driver specified.  Should there be?

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

Axtel4, 2016-08-14:
It looks like TP-Link was removed as a supported adaptor on December 2015.  https://github.com/cyoung/stratux/commit/f315ca5799ee868462d6a7e4af13b7d8c4097a29.  Referenced issue: https://github.com/cyoung/stratux/issues/119

Nokomis449, 2016-08-15:
Thanks.  I'm the guy who was using a TP-Link because iFly's flightplan sharing wouldn't work for me using the EdiMAX.  The TP-Link worked fine up until I upgraded to 1r0 last night (I jumped from 9x1).  So in the non-EdiMAX hostapd.conf file, shouldn't there still be a driver listed?  And do you know what it is/was?  I've overwritten those old SD's and don't know how to look thru github to find an old hostapd.conf file.

Nokomis449, 2016-08-15:
Nevermind, I found the old files and the non-EdiMAX .conf file doesn't reference a driver either.

closed 2016-08-15 by Nokomis449