cyoung/stratux #476: Evaluate "ownship traffic suppression" feature

cyoung, 2016-08-16:
Stratux version: v0.9b2

How well is it working?

Ergonomicmike, 2016-08-17:
I'm confused. IFly has its own ownship suppression algorithm. How does Stratux's work? Is Stratux's suppression only for Mode S with hex?  Does it identify ownship for EFB's and set a bit? 

I'm Mode C only. Does this test apply to me?

cyoung, 2016-08-24:
Only applies for ADS-B Out users that have put their hex code into the "settings" page.

Ergonomicmike, 2016-08-25:
Okay. Tnx for the clarification.

ssokol, 2016-12-02:
At this point, if I understand the code in traffic.go, we're excluding all traffic messages that match the ICAO code entered for "OwnshipModeS". Those messages get assigned to a variable (OwnshipTrafficInfo) but we never do anything with it.

What if, when we detected these ownship messages, we disabled the generation of our own ownship messages in gen_gdl90.go and instead swapped the message identifier byte to 0x0A (ownship) and forwarded them along to the app? This would eliminate the occasional and rather confusing "Stratux" callsign from appearing.

cyoung, 2016-12-02:
Okay, give this a try: 

https://github.com/cyoung/stratux/commit/9e8d20e3da53ab2974ec6400dcc34ca93e0c0f3d

http://updates.stratux.me/builds/update-stratux-v1.0r1-use_modes-9e8d20e3da.sh

nibake, 2017-04-01:
Hello all. I have a suggestion that may be best placed on this thread. I'm looking for ways to improve the ability to eliminate the ownship ghost.

Originally, I thought it would be great to be able to add multiple Mode S codes on the settings page, and select the airplane I am currently in. (I switch back and forth between about 6 different airplanes on a daily basis.)

I soon learned that TIS-B doesn't use Mode S codes and generates a random code for aircraft that aren't ADS-B out, so the previous solution, which may be helpful for some, would not have worked for me, since all of our airplanes are Mode S.  

Another solution, which would probably be more universal, would be to simply make it so the user can select the problem aircraft on the traffic screen and choose to block it for the rest of the sight. This may seem like a small thing, since one could take the code and enter it in the Mode S field on the settings screen, but in the environment in which I fly (never a boring moment), the extra efficiency would be very helpful. 

https://www.reddit.com/r/stratux/comments/623lob/options_for_mode_s_entry/

cyoung, 2017-04-19:
>I soon learned that TIS-B doesn't use Mode S codes and generates a random code for aircraft that aren't ADS-B out, so the previous solution, which may be helpful for some, would not have worked for me, since all of our airplanes are Mode S.

Not clear on this. Your TIS-B broadcasted target of ownship should be marked with your same Mode S code, if you are broadcasting Mode S.

> Another solution, which would probably be more universal, would be to simply make it so the user can select the problem aircraft on the traffic screen and choose to block it for the rest of the [...]

Good idea. I think we will implement this.