cyoung/stratux #477: Evaluate PingEFB performance

cyoung, 2016-08-16:
Stratux version: v1.0r1-5a13cc01ee

http://updates.stratux.me/builds/update-stratux-v1.0r1-5a13cc01ee.sh

1) How well is the hardware working?
2) How well is the Stratux integration working?

vortexjm, 2016-08-20:
Running Version: v1.0r1 (5a13cc01ee) witha PingEFP purchased at AirVenture.

I am questioning whether my 978 frequency band radio is working correctly on my PingEFB.  I've flown with it around the area a couple of times his week into areas where I had a lot of success using my SDR radios and crappy antennas.  I receive a ton of 1090 traffic in the FL levels (even from my basement) but never see towers/weather or 978 traffic.  I even had a buddy do a flyby about 2 miles from my home with his ads-b out equipped plane, but nothing received in the traffic status screen of Stratux or my EFB software.

cyoung, 2016-08-20:
@vortexjm - are you getting any 978 messages at any altitude?

Check the "max" number on the web interface http://192.168.10.1/

tc0nn, 2016-08-20:
I'm having the same issue. Brand new install though, so I'm in the learning curve. Don't leave Verbose logging turned on without a decent sized SD and don't forget to expand the filesystem... This will cause all sorts of madness when the SD partition hits 100%.

vortexjm, 2016-08-21:
I don't recall seeing any 978 messages, but will try to verify in the air
tomorrow.  I live in a bit of an ads-b tower dead spot (central Wisconsin),
but know where the edge of coverage normally is.

alexaggie, 2016-08-21:
@cyoung are there more changes in this one past what is in the e3d348a0fa release?

cyoung, 2016-08-21:
@alexaggie - no.

aerokinetic, 2016-08-22:
My PingEFB seems to be working fine. It was purchased at OSH. 
I've only flown with it once and the testing is quite unscientific. Over Indianapolis area at 2000'MSL (1200'AGL'ish) it was picking up the usual 2 ground stations and loads of traffic. Both 1090 and 978 *seemed* to be bringing in at least as many (if not more) messages as I am used to getting with the stratux dedicated SDRs and Dan Murray high-gain antennae.

cyoung, 2016-08-24:
Excellent, thanks for the feedback.

glongpilot, 2016-08-28:
I just downloaded the latest file from stratux.me v1.0r1 (945c8bf6bc) as reported on status tab and the PingEFB appears to not be supported (IE: no choice in the settings tab).

cyoung, 2016-08-28:
@glongpilot: update with this file via Settings: http://updates.stratux.me/builds/update-stratux-v1.0r1-5a13cc01ee.sh

Make sure to refresh the browser cache for the added webui settings.

EasternPA, 2017-02-24:
Old thread, but it's still open. I'm on my fifth day with my PingEFB on 1.2r1. It was a plug and play drop-in replacement. I'm a road warrior, not a pilot so I'm just an enthusiast. I am able to receive the one tower at KTTN from home. My current assignment has me working at KDAL (0 towers) and staying near KDFW (4 towers, I normally receive 1-3). The performance is excellent, and the mobility of the package just went through the roof. I keep the PingEFB in an empty prescription pill bottle to protect it from damage while being tossed around in my laptop bag.

The only verifiable concern I have right now is that all 1090 traffic is shown as -50.00 dB of power. I'm thinking this could have implications if I decide to use the PingEFB for 1090 reporting to FlightAware, FR24, or others. Don't they use that power figure as part of the MLAT calculations to create a synthetic position for what Stratux shows as "Basic Mode S and No-Position Messages"? That's likely not a Stratux issue unless the 1090 power level is being reported somewhere else.

Again, I have been very impressed with the performance as a drop-in replacement. It lets me enjoy my hobby during the week instead of trying to jam in a few hours during my short weekends at home.

Pi 2B, Stratux 1.2r1, Avare on Nexus 6 and Asus 7" tablet, FltPlan Go on iPad 2 (no internal GPS).

cyoung, 2017-03-13:
@EasternPA - thanks!

I'm closing this now since there isn't much happening with the PingEFB since we added support.

closed 2017-03-13 by cyoung

EasternPA, 2017-03-13:
Great job integrating this, and it's already been explained to me that the uniform signal strength for all 1090 traffic is not a cause for concern. Thanks again.