cyoung/stratux #485: iFly might not be accepting buffered FIS-B data

Ergonomicmike, 2016-08-29:
1. Stratux version: 1.0r1
2. Stratux config:

    SDR
      - [ ] single
      - [X] dual

    GPS
      - [X] yes
      - [ ] no
      type: BU-353

    AHRS
      - [ ] yes
      - [X] no

    power source:

    usb cable:

3. EFB app and version: iFly v9.7 beta

    EFB platform: Android

    EFB hardware: tablets

4. Description of your issue:

I'm guessing that this is more an issue with iFly than it is with Stratux. But, per my 6 minute video, it appears that iFly is not accepting buffered FIS-B data when it wakes up from sleep. (Or it might be accepting buffered METAR data, but not NEXRAD data.)

https://vid.me/fvKK

In previous testing in issue #433, I've observed that the message queue in Stratux builds up when it detects that iFly is sleeping. And then when iFly wakes up, the queue goes to zero, indicating that the buffered data is being sent to the EFB.

Nevertheless, it takes the same amount of time (2 to 3 minutes) for iFly to paint a new NEXRAD image from "buffered" data as it does in a cold start of both iFly and Stratux. This leads me to believe that iFly is not using the buffered NEXRAD data (or that Stratux does not buffer NEXRAD data) and that the 2 to 3 minutes is simply due to the time it takes for a new image to come in over the air.

This was a nice controlled test in that I was on the ground with an ADS-B tower nearby; it was night, with low traffic in the area; and there was weather to paint. (Given the last two, it seemed that FIS-B NEXRAD was being updated every few minutes.)

I would be interested in hearing from @cyoung if Stratux buffers NEXRAD. And if it does, I would be interesting in hearing (or seeing) if a different EFB accepts buffered NEXRAD data.

Ergonomicmike, 2016-08-29:
P.S. Does anyone know what the checklist icon by my username (currently showing "3 of 6") means? (It shows on the main issues page for this issue.) I searched for an hour and can't find the answer.

JohnOCFII, 2016-08-29:
> P.S. Does anyone know what the checklist icon by my username (currently showing "3 of 6") means? (It shows on the main issues page for this issue.) I searched for an hour and can't find the answer.

I'm not seeing it.  I wonder if only you can see that message.

Ergonomicmike, 2016-08-29:
I see it whether I'm logged or not. I also see it on someone else's issue, #473.

Ergonomicmike, 2016-09-05:
Question: Does FIS-B buffering work for anyone with a different EFB? (I suppose I should test w/ Avare the next time the opportunity presents itself.) And a related question: when the Stratux is receiving FIS-B data, and when the Stratux detects that the EFB has woken up, does the latest FIS-B data still go to end of the buffer? And when the buffer is full, I assume the First In gets deleted?

Ergonomicmike, 2016-09-29:
IFly found that it wasn't processing NEXRAD data properly. I flew 6 hours that last couple days with convective activity around and it looks like iFly has things working better. So it might handle buffered NEXRAD data better too. Will report the next time I'm able to test on the ground.

Ergonomicmike, 2016-10-10:
I'm closing this. I flew to KPAN the other day, where one can receive ADS-B on the ground. I put my tablet to sleep while I ate dinner. When I came back to the plane and woke the tabet up, NEXRAD updated within a minute. This was with iFly 9.7.49b and the latest version of Stratux. I know that iFly fixed a NEXRAD problem recently - I don't think Stratux had a problem and I don't recall seeing any commits about this. So perhaps an external bug all this time.

closed 2016-10-10 by Ergonomicmike

jdwall, 2016-12-03:
@Ergonomicmike and @JohnOCFII, just got around to seeing you all talking about the checklist piece in the message. See below for the syntax example. Github expects each `- [ ]` to be an item needing to be done. 

```
2) GPS
- [X] Yes
- [ ] No
```

Renders as:

2) GPS
- [X] Yes
- [ ] No

Now, on the issues page it shows the number of them completed from the first post.

[image]