cyoung/stratux #493: I combined the fan control and screen control scripts and tweaked the display

Nokomis449, 2016-09-29:
The screen now cycles thru Weather, Traffic, GPS, and Temp screens.  The Weather screen also shows the tower count, and the Temp screen shows the fan status (Off, Low, Med, Hi).  The fan isn't checked in the first 30ish seconds (shows "Startup"), and is then checked every 30ish seconds.  if the temp is above 110ish F it kicks on at 50% PWM; above 120ish is 75% PWM, and above 130ish is 100% PWM.  The pages cycle every 3 seconds.  I took the liberty to change "UAT" to "Weather" and "1090" to "Traffic" because that's how I view them.

Testing at home shows most everything is working as expected, but I haven't tested every aspect.  For instance, the larger font may need to be adjusted when the "Current/Max" numbers climb into the 10,000's.  

Be sure to remove or comment out the fancontrol.py line in rc.local so that two scripts aren't trying to control the fan.

The purpose of the larger font is to let my old eyes read the screen from a distance.  Now I'm going to work on remoting the screen to the panel while the Stratux resides on the hatrack in the back.

Comments welcome.

[stratux-screen.zip](https://github.com/cyoung/stratux/files/499698/stratux-screen.zip)

Nokomis449, 2016-09-29:
Perhaps this was not the best way to share the code, but I wasn't sure how else to do it.  So since it's not really an "issue", I'll close it.

cyoung, 2016-09-29:
Thanks for this. Let's leave the issue open for comments if anyone else wants to test it out.

Nokomis449, 2016-09-29:
I'd like to add the "# of targets seen" to the traffic screen, but couldn't figure out how to pull that info.  Also, it looks like the aircraft list might be sorted by distance.  If so & I can pull the closest one's distance, I'd like to quickly flash the Stratux logo any time a target is within, say, 1 mile.

Nokomis449, 2016-09-29:
Here's a video of the screen in action.  Notice the fan stops when the CPU temp drops below 110F, and the fan message goes from Low to Off.

https://youtu.be/ueuMbuCjAWQ

Nokomis449, 2016-10-02:
Screen lines are now centered and spacing is better.  I put a placeholder on the Traffic screen to display # of targets, but someone will have to point me to an example of how to pull that number from Stratux.  The GPS status says "Disconnected" when there's no GPS attached, is there any way to say the same thing if there's no UAT or 1090 dongle found?  As it is, it just reports 0/0.
Also, when a shutdown command is given, the screen stays lit, just like the red power light does.  Is there any way to tell the screen daemon to shut it down?

[stratux-screen (2).zip](https://github.com/cyoung/stratux/files/504735/stratux-screen.2.zip)

Nokomis449, 2016-10-02:
I know that only a few people have the screen, so here's a video of the latest version with all lines of data centered on the screen.

https://youtu.be/xDFBdEwBLEs

Nokomis449, 2016-10-11:
[stratux-screen.py.zip](https://github.com/cyoung/stratux/files/523014/stratux-screen.py.zip)

Here's my latest screen/fan script.  Still have some testing and work to do but I'd like comments if anyone has a screen.  The fan controller initiates at 100% for about 30 seconds, then throttles down to the appropriate PWM for the current temp.  It stays on all the time, but throttles down to 25% PWM below 110 degrees F.  The screen cycles through 6 pages: UAT, ES, Aircraft seen, GPS, Temp/fan, and Version/free disk space.  If free space drops below 0.5GB, it displays. "** LOW **".  It checks to see if UAT, ES, and GPS are turned on in the webpage, and does not show their screens if they are off.  As you toggle them on/off, their screens show accordingly.  Still need to determine if a fan is actually attached so I can skip that part if no fan, and need to determine if a screen is actually attached so I can skip the display code (and only run the fan, or kill myself if neither is attached).

skypuppy, 2016-10-11:
I recommend not disabling the screens, ever.  Simply show the device is off if it is.  More consistent UI and let's newbies know if they are missing some hardware if they are.  Also, showing them as off, if you think their on, is a good first step in hardware debugging.

Nokomis449, 2016-10-12:
I see your point @skypuppy, and here's my thinking.  I initially showed them all, all the time.  But as the screens matured and grew, it took longer to cycle thru them all to get back to the one I was interested in seeing.  Each of the 6 screens are displayed for 3.5 seconds.  So that's 21 seconds to loop through them all.  For someone with UAT only, it would still loop through the others even though they have no plans to add those.  My second point is that they'll need to be familiar with the 192.168.10.1 status screen to understand what they're seeing.  For instance, the UAT screen has 3 lines:  "UAT", "1024/1582", and "Towers: 4".  Without an understanding of the webpage status screen, the "1024/1582" is probably going to be confusing to a new user.  So there's a small learning curve, satisfied by knowing how to read the webpage.  I think if info from the little screen is expected but missing, the first place to check is intuitively going to be the webpage.

While thinking through my response, I've realized an oversight - I'm only showing the Temp if a fan is attached.  I need to display the temp even if (ESPECIALLY if) there's no fan.  So that's my next addition, to show the Temp screen even if there's no fan.  Also, (I think) I can show a message if there's two radios attached but only one turned ON in the webpage - that would be enough to put someone into headscratch mode.

skypuppy, 2016-10-12:
Oh.  I assumed pages were reached by user toggle, and not auto-paging.  Myself, I prefer user toggle.  If I'm busy flying (or doing anything else) I'm going to miss those screens anyway if they are auto-.

Nokomis449, 2016-10-12:
Agreed, but this little screen is a one-way communicator.  I'm probably the only person with it remote mounted/velcro'd on the panel.  I used CAT-5/RJ-45 and the screen only uses 4 wires, so that leaves me with 4 wires to play with.  I thought about adding some buttons, and a button to start/stop the scrolling was one of the ideas.

Another idea is to write a similar, more robust web page with toggles and buttons.  I have an old, small Motorola RAZR Android phone that would be perfect to replace this screen, with the benefit of color, two-way touchscreen communication, and no wires required.

Nokomis449, 2016-10-21:
Latest screen/fan script with commented code.
[screenfan.py 10.21.2016.zip](https://github.com/cyoung/stratux/files/545307/screenfan.py.10.21.2016.zip)

Video of screen in action:
https://youtu.be/F7VNVcsx3_o

The scrolling can be easily turned off if it's distracting.  The speed of the scrolling can be adjusted, and the amount of time the screens are displayed can be adjusted.  These adjustments (and the presence of a fan) are set in the first 4 lines of code.  The screen is autodetected; if there's no screen but there's a fan then only the fan code executes and vice versa.  The screens auto-adjust to reflect changes on the Stratux webpage such as UAT, GPS, etc.  For instance, if a radio gets turned off that radio's screen is no longer shown.  Obviously more people have fans than screens, so the fan flag defaults to True.  To run automatically, put "/usr/bin/screenfan.py start" in /etc/rc.local, and comment out the current fan and screen scripts.
Feedback appreciated!

Nokomis449, 2016-11-09:
[screenfan 11-8.py.zip](https://github.com/cyoung/stratux/files/580006/screenfan.11-8.py.zip)

Here's the latest version.  Includes the original (@cyoung 's) graphical screen which can be set as the only screen, in addition to the other screens, or hidden.
Other enhancements: the actual screen painting is now in a single function called by each screen, which reduced the lines of code by 50% and makes the code easier to follow.  Also added more comments in the code.
I'm guessing no one with a fan but no screen has tested this, because a serious bug existed which prevented the code from running if a screen was not attached (my testing at that time simply eliminated the screen via a software switch and it acted fine, but when the screen was physically disconnected the bug appeared),  Check the first few lines of code to set the switches the way you want.  I have tested most but not all switch scenario's.  Still could use some tidying up with the code as I learn more about Python.  Comments appreciated.

closed 2019-01-17 by Helno