cyoung/stratux #500: iPad Air 2 / iOS 10 lease detection issue

cyoung, 2016-10-01:
1. Stratux version: v1.0r1
2. Stratux config:

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [x] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [x] no

    power source: ?

    usb cable: ?

3. EFB app and version: N/A

    EFB platform: iOS 10

    EFB hardware: iPad Air 2

4. Description of your issue:

User reports that his Stratux build works fine on an old iPad, but a brand new iPad Air 2 doesn't show Stratux in the "Devices" list on FF.

User sent screenshots showing "Recent Clients: 0" from the iPad.
[image]
[image]
[image]

cyoung, 2016-10-03:
Issue wasn't related to the iPad.
Old iPad was working but new one wasn't. Old iPad was in DHCP lease file, new one wasn't. SD card had a write issue, so Stratux wasn't picking up the new DHCP lease. Need a UI error thrown when `/var/lib/dhcp/dhcpd.leases` is not writeable.

emuyshondt, 2016-10-03:
Here's what happened to me. I'm running an iPad Air 2 with the latest ForeFlight. I upgraded to iOS10, and I think also to the latest version of FF since I last used my FlightBox. I have the latest released version from both iOS10 and FF. I have been using my iPad with Stratux and FF for months. It worked before the latest software upgrades without issue.

I finally got to fly yesterday and connect to my ADS-B receiver. I had seen this issue report. When I first started up my iPad, I verified it was connected to the FlightBox (stratux derivative) WiFi network. I then went to the FF page described above and had the exact same symptom.  I clicked on the small "i" on the upper right hand corner and it showed the Stratux WiFi network. I tapped on that line, and then I got the message that shows something about ADS-B device connected. Everything worked as expected after that.

Now I don't know if this is a Stratux, FF or iOS 10 issue, since I updated both iOS and FF since I last used my setup.

cyoung, 2016-10-03:
@emuyshondt - Try disabling Bluetooth and see if it has any effect.

emuyshondt, 2016-10-03:
I didn't think to check if the problem re-occurred after the system started
working. I just flew and enjoyed having traffic information (two planes I
would have never seen without Straux within half a mile!). Unfortunately,
the Pi I have with me seems to be having problems, so I can't check until I
get to my plane again.

Now I'm going to be away from my plane and Stratux until the end of
October. If I get a chance, I'll try to go by the airport before I leave on
Thursday to try some experiments.

cyoung, 2016-11-17:
>SD card had a write issue, so Stratux wasn't picking up the new DHCP lease.

Solution was to replace SD card.

closed 2016-11-17 by cyoung