cyoung/stratux #502: Serial GPS(RY835AI) not reconnecting

peepsnet, 2016-10-02:
1. Stratux version:
2. Stratux config:

    SDR
      - [ ] single
      - [ X ] dual

    GPS
      - [ X ] yes
      - [ ] no
      type:

    AHRS
      - [ X ] yes
      - [ ] no

    power source:
ANKER 20800 and 3amp wallwart
    usb cable:
22awg portapow

3. EFB app and version: (e.g., WingX Pro7 8.6.2)
IFLY
    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

The GPS(RY835AI) will not reconnect after "service stratux stop | service stratux start" or switching the GPS off and on again in the webUI. 

Only a full restart will bring the GPS back!!

AvSquirrel, 2016-10-02:
See last comment in https://github.com/cyoung/stratux/issues/400. Fixed with https://github.com/cyoung/stratux/pull/464.

AvSquirrel, 2016-10-02:
@peepsnet --  `/boot/cmdline.txt` assigned /dev/ttyAMA0 as a console device on recent images. This will cause the serial connection to time out.

#464 updated the .sh installer scripts, but the makefile doesn't touch `/boot/cmdline.txt`... in other words, if you're building from source and `make install`ing, this might still be an issue.

TL;DR: Open `/boot/cmdline.txt` and see if it contains `console=ttyAMA0,115200`. If so, delete that text, save, and reboot. See if that resolved the connectivity issues.

peepsnet, 2016-10-02:
/boot/cmdline.txt contains

dwc_otg.lpm_enable=0
console=tty1
root=/dev/mmcblk0p2
rootfstype=ext4
elevator=deadline
fsck.repair=yes
rootwait

peepsnet, 2016-10-02:
OK now its not doing it!!!

closed 2016-10-11 by peepsnet