cyoung/stratux #507: Nano 2 disconnecting

boa888, 2016-10-07:
1. Stratux version:  stratux-v1.0r1-945c8bf6bc
2. Stratux config:

    SDR
      - [ x] single
      - [ ] dual

    GPS
      - [ x] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [x ] no

    power source: TSO'd regulated power supply

    usb cable: Anker 3amp

3. EFB app and version: (iFlyGPS)

    EFB platform: (Andorid)

    EFB hardware: (Samsung Galaxy Tab 2)

4. Description of your issue:
I keep getting disconnected from the FIS-B data stream. I have had to power cycle the Stratux to get it to re-connect during flight. Turns out this is not the problem. Through trial and frustration I have discovered that it is the NANO 2 that is disconnecting. All I have to do is unpplug the Nano and immediately plug it back in (un-plug time is not a factor) and it starts in a second or two. The Nano gets pretty warm so I stuck a heat sink on it but the problem persists. I moved the Nano right in front of the air vent. That kept it cool but the problem persists. 

I turned off the logs and deleted them. I will reinstate logging and post it (if I can figure out how)

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

Ergonomicmike, 2016-10-07:
Have you tried plugging the Nano into a different USB port? If that doesn't fix it, it might be a bad Nano. (You didn't say what type GPS you're running. If non-standard, perhaps a conflict?)

boa888, 2016-10-11:
I will try a different port later. GPS U-Blox VK-172 or internal GPS makes no difference

boa888, 2016-10-12:
Wild mouse syndrome.  I closed this by mistake.

Nokomis449, 2016-11-15:
I see you have a TSO'd power supply and a 3A cable, but I get the same symptoms when I connect to an inferior power supply.  Perhaps your TSO'd unit is rated for only one amp, or the cable has an internal fray.  My setup is: 12v ship's power plug with WalMart 5v adapter connected to/charging a Jackery 5v battery  which is connected to a remote controlled on/off relay which powers the Pi on the hatrack in back.  I flew several hours Friday with no such issue.  On the return trip I was in a hurry to get the plane loaded and just plugged ship's power directly to the Pi.  The result was that I lost UAT while everything else (ES, GPS, wifi) soldiered on.  A reboot (all I could do from the front was to cycle the power to the plug) fixed it for another 15-30 minutes, then rinse and repeat.

boa888, 2016-11-16:
Some say 'if it ain't broke don't fix it' but wish it would break so I can fix it.  The Stratux has not failed in quite some time now. I deleted the logs and turned of logging. If memory serves me it did fail after that but is has not given any problem in the last 20 hours. BUT I do NOT believe the logs had anything to do with the problem. The power supply is a 10A and I do believe it is working fine. It is possible that the power supply is noisy. My Stratux would run find for an hour or so then like yours' I had to unplug-plug every 10-15 min. If it fails again I will post.

Ergonomicmike, 2016-11-19:
Logging causes failures. https://github.com/cyoung/stratux/issues/449

cyoung, 2016-12-01:
> I have had to power cycle the Stratux to get it to re-connect during flight. Turns out this is not the problem. Through trial and frustration I have discovered that it is the NANO 2 that is disconnecting. 

So did power cycling it have the same effect as unplugging the SDR and re-inserting?

cyoung, 2017-04-19:
This sounds a lot like the issue addressed in #568. @boa888 - please give v1.2r2 a try, and if you continue to have this issue then please reopen or comment here. Thank you.

closed 2017-04-19 by cyoung