cyoung/stratux #509: I cannot re-download after upgrading iPad to iOS 9.3.5

tonyrizzo, 2016-10-07:
1. Stratux version: I don't know
2. Stratux config: I don't know

    SDR
      - [ ] single
      - [ ] dual

    GPS
      - [ ] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [ ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

tonyrizzo, 2016-10-08:
I had Stratux on my iPad 2 and was using it without problems with the unit I built in my Velocity.  When I upgraded my iPad to iOS 9.3.5 a problem occurred in the iPad that Apple could not solve without advising telling me to wipe the iPad and return it to the original factory settings.  Since then I cannot re-download Stratux.  Is there a compatibility problem with iOS 9.3.5?

Helno, 2016-10-09:
You really need to explain the issue you are having better. There is nothing to download onto your ipad to make Stratux work. If you want an icon for the Stratux web interface on your home screen you simply connect to the Stratux wifi network then open 192.168.10.1 in Safari, if it doesnt ask you to create a shortcut you can just manually create one by clicking the box with an up arrow then on the add to home screen button.

A home screen link does not effect functionality of stratux. 

Perhaps responding to the comments on slack would get this figured out in a few minutes.

cyoung, 2016-10-11:
@tonyrizzo - no compatibility issues known with iOS 9.3.5. Did you go to your "Settings" menu and connect to the "stratux" Wi-Fi network again?

tonyrizzo, 2016-10-11:
Thank you very much! Sadly, situation here has caused me to not be able to get to the computer for several days. I assume from your comment there were many responses on slack and I appreciate all of them. I will try what you just suggested.
Again, many thanks!
Tony 

Anthony M. Rizzo, M.D.
Colonel (Ret.), USAF, MC, SFS
Professor of Anatomy and Physiology
Polk State College

tonyrizzo, 2016-10-15:
I was finally able to get to the plane today, followed your advice, and all worked well. Many thanks!
Tony 

Anthony M. Rizzo, M.D.
Colonel (Ret.), USAF, MC, SFS
Professor of Anatomy and Physiology
Polk State College

cyoung, 2016-10-15:
Thanks for following up, I'll close this now.

closed 2016-10-15 by cyoung