cyoung/stratux #511: All traffic target source labeled as "978 TIS-B" in ForeFlight

cyoung, 2016-10-18:
1. Stratux version: v1.0r1
2. Stratux config:

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [x] yes
      - [ ] no
      type:

    power source: any

    usb cable: any

3. EFB app and version: ForeFlight (all tested)

    EFB platform: (any)

    EFB hardware: (any iPad, not sure how displayed on iPhone)

4. Description of your issue:

See screenshot.

1. How is ForeFlight determining the traffic source?
2. How can we send the traffic source?

@bsdcat - thoughts?

[image]

bsdcat, 2016-10-18:
Right now we're guessing, basically. GDL90 traffic reports don't include whether the report was picked up on 978 or 1090, and `makeTrafficReportMsg()` doesn't even reference Last_source.

I don't have any good ideas on where to try to stash it in the report, either. It looks like it's available via the `/traffic` endpoint, but I can't commit to adding support for another data stream at the moment.

cyoung, 2016-10-18:
We could track last source if we were to send it somehow. The tail number is modified with a 'u' and 'e' in front of it when `DisplayTrafficSource` is enabled - but that reduces the field size by one of course.

See http://www.faa.gov/nextgen/programs/adsb/wsa/media/GDL90_Public_ICD_RevA.PDF
"3.5.1.2  TARGETIDENTITY"

The high bits on this field are unused. Are you using these bits for anything? We could use this for traffic source.

Are you obtaining the traffic source for any GDL90 data stream? If so, could we use whatever method you're using now?

If unable to make any of those changes, would it be possible to remove it? The only time I hear about it is when people are confused by it.

bsdcat, 2016-10-18:
I don't have an answer right now, but I'm tracking this internally and figuring out what exactly we can do.

AvSquirrel, 2016-10-19:
@cyoung - The last half byte of the GDL90 traffic report message is defined as "Spare (reserved for future use)". I'd suggest that one or two of these unused bits could be redefined to show traffic source.

[image]

Justification:

1. This is a legacy specification; it is unlikely that its owner (Garmin) will revise it at any point in the future.
2. Theoretically, other GDL90-over-WiFi boxes on the market aren't using those bits (maybe @bsdcat could verify), and theoretically, "unaware" software should ignore those bits. (Verify.)
3.  It would avoid the need to create auxiliary message types, or to rewrite EFB software to parse the /traffic websocket.
4. It's the future.

For Stratux as it's written today, you could get away with using a single bit (e.g. 0 = 978 MHz, 1 = 1090 MHz). However, if there's a reasonable chance that Stratux could gain the ability to take traffic inputs from other sources (ZAON PCAS, FLARM, APRS, etc) either directly or via plugin / server architecture at some point in the future. I'd just grab two bits right away to avoid having to redefine this again.

Such an implementation could look like this:

[image]

**3.5.1.13 TRAFFIC SOURCE**
The Traffic Source is a 2-bit value "f" that provides status information about the source of the traffic message. The values for "f" are given below:
f = 0 : 978 MHz ADS-B / ADS-R / TIS-B
f = 1 : 1090 MHz ADS-B / ADS-R / TIS-B
f = 2 : FLARM receiver
f = 3 : Other (APRS, ZAON PCAS, etc.)

cyoung, 2016-10-19:
GDL90 will always be second choice to providing data to an EFB via JSON feed.

I'm in to try the proposed changes, but the ideal case would be to leave GDL90 alone and switch to a JSON feed for additions like these.

It was surprising to see on some panel avionics what Stratux GDL90 over serial output does. One thing that was funny was seeing how the "alert" bit was handled differently from any other EFB. I'd rather tailor GDL90 modifications for those cases versus EFBs which have a faster update cycle and have the resources available.

cyoung, 2016-12-01:
Stale issue, nothing further we can do here.

closed 2016-12-01 by cyoung