cyoung/stratux #517: synthetic vision

9foxlabs, 2016-11-06:
1. Stratux version:
2. Stratux config:

    SDR
      - [ ] single
      - [ ] dual

    GPS
      - [ ] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [ ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue: No issue, but when do you think synthetic vision will be available? It's one of my primary reasons for purchasing the name neanderthal one. I do appreciate the open source of it though

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

9foxlabs, 2016-11-06:
"Purchasing the name brand one", not "neanderthal". Lol

Ergonomicmike, 2016-11-09:
Synthetic Vision is a function of the EFB. (Not the Stratux, per se.)

cyoung, 2016-11-16:
@9foxlabs - the neanderthal AHRS are more advanced than ours at the moment. The terrain portion of synthetic vision does not require AHRS, however, just a GPS position. You could use synthetic vision for obstacles using stratux as a GPS source, just no attitude reference.

closed 2016-11-16 by cyoung