cyoung/stratux #518: Write daemon to manage GPIO accessories

andofrjando, 2016-11-07:
As a bit of an extension to Issue #493, I've been considering writing a daemon to manage all accessories connected to the GPIO pins (eg OLED screen, fan control, safe shutdown switch). Some of the code already exists in the repo, but thought it would be neater to repackage it all into a single daemon that attempts to detect what's connected and only run necessary code for those accessories. The only exception might be the PWM fan control that could probably be toggled in the settings web page. 

The advantage being that GPIO components would be plug and play for users and all the GPIO relevant code would be in one place. There would just need to be some documentation on wiring connections (for example how to wire a shutdown button with pull-up/pull-down resistors). 

This would probably to me a few weeks to put together in Python (still waiting for my hardware in the post), so I'm wondering if anyone thinks this would be a worthwhile contribution?

jamez70, 2016-11-14:
On a few of my designs I added custom LEDs to the GPIO connector, and I usually update the code to set/reset those LEDs. THis may be a good way to handle that

cyoung, 2016-11-16:
@andofrjando - definitely worthwhile, we need some nice way to package this all together.

If you're able to write something in there where the stratux daemon becomes aware of the presence of each one of these devices, that would be great.

jamez70, 2016-11-16:
I had an idea where at different points there would be 'triggers', for instance, a trigger would be reception of a message on 978mhz. There would be a map that had different trigger points would perform some activity on a gpio (on, off, toggle, etc).

There would be a UI that would list the GPIOs, and a drop down box or checkbox that you could assign one or more triggers to the event.

andofrjando, 2016-11-18:
Ive got something half finished, but will be busy for the next week or so. At the moment the only sub modules I could think of are one to control a PWM fan, one for a OLED screen controller and one for a simple push button safe shut down. 
Another idea that I plan to add to my own setup is a battery level monitor to be used with a power boost 1000c, but I don't have it (or a Pi zero) yet. 

The daemon is written in python to take advantage of of the excellent GPIO package written for the Pi.

ssokol, 2016-12-02:
It's specific to the AHRS board, but here's a start at a status monitoring daemon:
https://github.com/ssokol/boardctl

closed 2020-03-19 by andofrjando