cyoung/stratux #530: Need help modifing the Tower dispay with Foreflight

767heavy, 2016-11-22:
Hi,
How do I modify the font color of the Towers?  The Red with the Black outline is sometimes difficult to see.
I'm new , also having a hard time on finding and accessing the file that contains the coding.  
(and what program do I use to modify the coding)

thanks.

AMNegron, 2016-11-26:
Pretty sure that is coded in FF or whatever EFB/display app you're using.

cyoung, 2016-12-01:
@767heavy - can't do it, send a message to FF support to suggest they change it if you think it's an issue.

closed 2016-12-01 by cyoung

767heavy, 2016-12-01:
Thanks for the reply,

I had one more question. What does adding the GPS antenna do for me? I'm
using the unit with ForeFlight on my Ipad which has a GPS  built in.
Do I need the extra GPS antenna?

thanks.

cyoung, 2016-12-01:
@767heavy - if you are happy with the iPad GPS performance, then it doesn't add much for you.

Are you flying a 767? GPS could add "pseudo-attitude". If you are indeed flying a 767 then I'd be curious to see your impressions on a fake attitude indicator based on GPS readings.

767heavy, 2016-12-01:
Yes I do. 
I fly the Boeing 757 and 767.
I plan on taking the unit with me on my next Atlantic crossing.
I'll let you know how it works out.

cyoung, 2016-12-01:
Send me a note at cyoung@gmail.com if you could, I can help you get set up with the GPS attitude update.

http://updates.stratux.me/builds/update-stratux-v1.0r1-ahrs-ebfb671142.sh