cyoung/stratux #534: Stratux v1.0r1 + FlyQ 2.4.3 no connection

jwbruce, 2016-11-26:
1. Stratux version: v1.0r1
2. Stratux config:

    SDR
      - [ ] single
      - [X ] dual

    GPS
      - [X ] yes
      - [ ] no
      type: gmouse USB

    AHRS
      - [ ] yes
      - [ X] no

    power source:  wall USB adapter

    usb cable:  quality

3. EFB app and version: (e.g., WingX Pro7 8.6.2)
FlyQ 2.4.3

    EFB platform: (e.g., iOS 9.2)
10.1.1
    EFB hardware: (e.g., iPad Mini 2)
iPad mini2

4. Description of your issue:
FlyQ never detects or acquires GPS or ADS-B from stratus.  Stratux webpage shows solid GPS lock and geometry with numerous 1090ES traffic being tracked.  FlyQ GPS indicator remains RED.  ADSB indicator is black.  Have turned location services ON and OFF.  No change.  The "ignore WifI GPS/ADSB" in FlyQ is also off..   I only use my Ipad EFB purposes in the plane so I am not an apple power users. (I use my android for general purpose tablet ops)

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

jwbruce, 2016-11-26:
OK. It's has to be me being stupid. FltPlan Go on my Android won't connect either.

What am I doing wrong?

cyoung, 2016-12-01:
http://192.168.10.1/, are you able to connect and view messages coming in?

jwbruce, 2016-12-02:
I see traffic "flying by"' on the traffic page at 10.1. I see the GPS position updating on the web page GPS tab too. But neither app -- FlyQ on my iPad mini and FltPlan Go on my Android -- seem to detect any stratux data, I.e. no GPS position and no traffic.

I am too far away from an adsb ground station to receive any TIS or FIS in the ground.

JW

N3740J, 2016-12-11:
I, too am having the same issue. I just got a new iPad Mini 4, running the same iOS and FF versions (latest as of 12/2016) and I am able to open the stratux menu from my browser but ForeFlight will not recognize it. Same versions on my other iPad and my iPhone, ForeFlight recognizes them fine.

The general info button under devices in FF alternate between wifi IP addresses of 192.168.1.13 and 192.168.1.15.

aputnam52, 2016-12-27:
I'm actually running into the exact same issue as N3740J...  New iPad mini and FF won't recognize the Stratux.  Two other iPads and an iPhone - no issues at all.  Seems odd, going to watch this thread closely...  192.168.10.1 connects nicely and everything looks perfectly normal, but FF shows no connected devices.

jwbruce, 2017-01-12:
Any progress on this issue?  My navigation apps on my iPad and my Android fail to see the stratux at all, even though the same tabs see the stratux working fine via the webpage.  So, the stratux is just a pretty brick for me right now.

I can troubleshoot on my devices if that would help.  I just will need some guidance from cyoung.   I am not familiar enough with the stratux system to know where to begin.

cyoung, 2017-01-17:
Could you take a screenshot of the main status page http://192.168.10.1/ and post it?

jwbruce, 2017-01-19:
I flashed the 1.0r1 image to another SD card.  (The new one is 16GB as opposed to the 8GB original one.)
Fired up the stratux, and FlyQ find the stratux now.

I haven't had time to reflash the original SD card to see if the card is bad, or the flashing was bad.

cyoung, 2017-01-19:
@jwbruce - thanks for following up.

@N3740J - could you try flashing another SD card to see if that's maybe it?

Ref #500.

There's a new warning in the latest version which will throw a webui error when the leases file is not writeable - doesn't fix this failure but at least alerts.

cyoung, 2017-03-13:
Sounds like a bad SD card still.

closed 2017-03-13 by cyoung