cyoung/stratux #536: 1.5 hour flight in heavy traffic, message queue overflow

roperf15, 2016-11-28:
1. Stratux version:1.0r1 (5c70640eac)
2. Stratux config:

    SDR
      - [ ] single
      - [X ] dual

    GPS
      - [X ] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [X ] no

    power source:

    usb cable: yes - ship's power

3. EFB app and version: (e.g., WingX Pro7 8.6.2)  FF latest

    EFB platform: (e.g., iOS 9.2) iOS 9.3

    EFB hardware: (e.g., iPad Mini 2) IPAD 2

4. Description of your issue: 
No external apparent issue.  I flew 2 flights (1.5 each) near NYC and Philly.  You can see in the logs that I began to get a message que overflow.  I'm not sure if it was due to the sheer number of aircraft that were around.  The same messages were logged on both flights.  Again, during the flight I wasn't aware of any issues.  Not sure if this is expected behavior or not.  Since I was using one of the latest commits, this was just something that I thought you guys might be interested in.  This cycle repeated over and over again in the logs.  If you need me to search for anything else in the logs and send it to you, let me know.

here is an excerpt from the log:

2016/11/28 00:04:41 stats [started: 23 minutes ago]
2016/11/28 00:04:41  - Disk bytes used = 1.9 GB (81.4 %), Disk bytes free = 442 MB (18.6 %)
2016/11/28 00:04:41  - CPUTemp=45.08 deg C, MemStats.Alloc=46 MB, MemStats.Sys=76 MB, totalNetworkMessagesSent=19,742
2016/11/28 00:04:41  - UAT/min 878/1,494 [maxSS=106.40%], ES/min 17,285/27,015, Total traffic targets tracked=602
2016/11/28 00:04:41  - Network data messages sent: 18408 total, 10461 nonqueueable.  Network data bytes sent: 22601944 total, 2147115 nonqueueable.
2016/11/28 00:04:41  - Last GPS fix: now, GPS solution type: 2 using 13 satellites (14/16 seen/tracked), NACp: 11, est accuracy 2.40 m
2016/11/28 00:04:41  - GPS vertical velocity: 2.94 ft/sec; GPS vertical accuracy: 2.2 m
2016/11/28 00:04:41 Ownship target detected for code A9XXXX
2016/11/28 00:04:42 Ownship target detected for code A9XXXX
2016/11/28 00:04:43 192.168.10.11:4000 - message queue overflow.
2016/11/28 00:04:43 192.168.10.10:4000 - message queue overflow.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

Ergonomicmike, 2016-11-29:
Duplicate issue?

https://github.com/cyoung/stratux/issues/449

cyoung, 2016-11-29:
@roperf15 - thanks for reporting. This is actually normal behavior. The "warning" sounds ominous, but it's more or less expected. Only weather data is stored in the "message queue". If you see the message counts, ~57% of your messages were unqueueable.

The message queue is of fixed size, so if you have a device or device(s) that haven't connected in a while (or even on this startup), the queue will overflow. It's typically large enough that this doesn't happen unless you leave your tablet/EFB shut down for more than 30 minutes. There usually isn't much consequence to it happening, either, as redundant data has usually been received by the time it overflows.

Take a look at `/var/lib/dhcp/dhcpd.leases` to see what .10 and .11 were, maybe they weren't even connected during the flight.

cyoung, 2016-11-29:
@Ergonomicmike - different issue. This has to do with the sleep mode FIS-B queue. I should probably remove this "warning" since it's more of a debugging message.

closed 2016-11-29 by cyoung

Nokomis449, 2016-11-29:
From the log, it also looks like @roperf15 didn't expand the microSD and the available disk space is getting low.