cyoung/stratux #540: Issue with ipad mini.

sebnad, 2016-12-06:
Hello, trying to connect stratux v1.0r1 on pi3 to FltPlan Go app using an ipad mini model A-1432 with IOS version 9.3.5. The stratux page is working fine but unable to connect stratux in FltPlan Go.. showing connecting... but nothing happennig. No ads-b and no gps connections

The stratux is working fine with ipad air and iphone 6..
Any suggestions?

1. Stratux version: v1.0r1
2. Stratux config:

    SDR
      - [ ] single
      - [yes ] dual

    GPS
      - [yes ] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [no ] no

    power source:power bank pb6000cb

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

cyoung, 2016-12-06:
External menu -> Stratux. This is where you see "connecting"? Do you have any other EFBs open?

sebnad, 2016-12-06:
No, the only other app is garmin pilot and i did make sure it was closed.

sebnad, 2016-12-06:
The conncecting promt is in the fltplan go app under external tab.

sebnad, 2016-12-06:
Here is the stratux working fine and the fltplan go app not connecting. 

[image]
[image]

FltPlan, 2016-12-07:
Hi sebnad,

We have had no known issues and no other complaints with FltPlan Go connecting to Stratux.
We were able to connect Stratux to FltPlan Go on both an iPad Mini and iPad Air 2 without issue.

Please make sure of the following:
- You do not have any other apps open and that Stratux is not in use with another app.
- You have the most current version of FltPlan Go from the Apple App Store.

You can try a hard reset of the iPad Mini (hold the iPad home button and lock button at the same time).
Additionally, you can try disconnecting and reconnecting to the Stratux unit.
   - iPad Settings - Wi-Fi - tap on the little (i) icon next to Stratux - Forget This Network and then reconnect.
   - FltPlan Go - External - Stratux.

If you need additional assistance, please send an email to Tech@FltPlan.com

Thank you,
Tech@FltPlan.com

bojon01, 2016-12-08:
I have the same issue with my new iPad Mini 4 from Verizon.  I have an older Mini that is working just fine.  I am thinking that port 4000 is possibly being used by the newer iPad Mini 4 already because I have tried everything I know to do.  I can see that the device is working when I go to the web page when connected to the Stratux WiFi from my iPad.  None of the EFBs work with the new iPad Mini 4 and my Stratux device.  I see quite a few comments about this on Google, but hoping I could get some help here.

bojon01, 2016-12-08:
I finally got this working by reimaging another SD card.  Not sure what happened to the other but I pull the most recent version (both appeared to the be same new release), but somehow maybe the SD was corrupted.   Good luck if you are having this issue as it was very frustrating.  Thanks to CYoung for this great software and my hat goes off to you CYoung!

sebnad, 2016-12-08:
Yes, same problem..except for me it is with a ipad mini.. i will try with a new sd card thank you!

cyoung, 2016-12-08:
https://github.com/cyoung/stratux/issues/500

sebnad, 2016-12-08:
I have installed stratux on a new sd card and now fltplan go connect to stratux with all my devices(ipad mini iphone 6 and ipad air) thank you!!!

closed 2016-12-09 by sebnad

cyoung, 2016-12-09:
One further note, make sure you check "Recent clients" first - if you're connect and it's at "0", that's a problem!

sebnad, 2016-12-10:
Yes, now everything is perfect,  just did a flight with 4 deviced connected to stratux and works A1! Thanks Cyoung😉