cyoung/stratux #541: Adafruit Ultimate GPS compatibility

johnpayne, 2016-12-09:
1. Stratux version: 1.0r1
2. Stratux config: Raspberry Pi 3 B

    SDR
      - [ ] single
      - [X] dual

    GPS
      - [X] yes
      - [ ] no
      type:  [Adafruit Ultimate GPS](https://www.adafruit.com/product/746)

    AHRS
      - [ ] yes
      - [X] no

    power source:  2.0v wall adapter

4. Description of your issue:
If this is a dead topic and you guys aren't interested in supporting the [Adafruit Ultimate GPS](https://www.adafruit.com/product/746), i understand and you can close this, but I think i'm close to finding the problem.  Just need some guidance.

I have the [Adafruit Ultimate GPS](https://www.adafruit.com/product/746) connected through serial to the GPIO.  I have noticed that even though the GPS led blinks once every 15 sec (indicating a 3D Fix) it intermittently shows connectivity on the stratux status web page.  I've tracked it down to what I think might be the system using the serial port for something else and blocking the GPS serial reader and causing stratux code to exit.  It takes quite a while for the GPS to come back up on the stratux status web page.

/boot/config.txt has the following changes:
dtoverlay=pi3-miniuart-bt
enable_uart=1

when watching the NMEA data stream through the serial interface, i notice it works for a few seconds and then starts sending non NMEA data and errors out (see attachment NMEA stream example.txt)

Am I headed down the right track?  Any Ideas?

[NMEA stream example.txt](https://github.com/cyoung/stratux/files/641593/NMEA.stream.example.txt)
[stratux.log.txt](https://github.com/cyoung/stratux/files/641594/stratux.log.txt)
[syslog.log.txt](https://github.com/cyoung/stratux/files/641592/syslog.log.txt)

cyoung, 2016-12-09:
Not much to add, but I think this might be related to https://github.com/cyoung/stratux/pull/505.

johnpayne, 2016-12-10:
It seems a bit different than [#505](https://github.com/cyoung/stratux/pull/505).  It's not just garbage mixed in with NMEA Data, it seems as though the NMEA data is blocked entirely by the garbage.  I'll have to check and make sure I don't have any bad connections.  It's pretty easy for me to reproduce.  Also, when the power is unplugged to the GPS unit, nothing comes through the serial port at all...so It's possible the garbage is either produced by the GPS itself, or coming from a bad connection.  I suppose it could also be a baud rate issue?

cyoung, 2016-12-10:
Try a USB-to-Serial dongle and compare results.

johnpayne, 2016-12-10:
Ok will do

johnpayne, 2016-12-12:
I don't have a USB to Serial dongle yet, but one thing I have noticed so far: 
NMEA streams beautifully through /dev/ttyS0.   However, when I swap serial ports (using dtoverlay=pi3-disable-bt) NMEA stream is interrupted and blocked every few seconds....very consistently.

There must be something else that tries to use ttyAMA0 when I swap ports.  I have turned the console off to serial port, and I have even tried stopping getty services for ttyAMA0 and ttyS0.  Nothing seems to make a difference.

[image]

johnpayne, 2016-12-13:
A test running off of port ttyS0 ran for almost 17 hours before i shut it down.  Unfortunately, stratux won't detect and use that port.  I'll keep trying to see if I can figure out what is blocking ttyAMA0.

[image]

johnpayne, 2016-12-14:
When i stop the Stratux service, the Blocking Error on ttyAMA0 goes away.

Seems like there is something in the Stratux service that is reading ttyAMA0 with a block.

johnpayne, 2016-12-21:
Under v1.1b1, the issue is still manifest.  Here is how it shows up:
Occasionally when the code invokes scanner.Scan() to read input from the serial, occasionally the buffer limit is exceeded and the gps reader exits:
```
2016/12/21 21:13:54 reading standard input: bufio.Scanner: token too long
2016/12/21 21:13:54 Exiting gpsSerialReader() after i=0 loops
```
I'm still trying to track down the source of the non NMEA data that is occasionally making it's way into the stream:
```
2016/12/21 01:07:17 Finished writing u-blox GPS config to /dev/ttyAMA0. Opening port to test connection.
2016/12/21 01:07:18 GPS error. Invalid NMEA string: Invalid NMEA message:
�隮H��
2016/12/21 01:07:18 GPS error. Invalid NMEA string: Invalid NMEA message:
u5eŢb�b��bN�b��b���bR��jR"�
2016/12/21 01:07:18 Couldn't calculate GPS-based attitude statistics
```
Any ideas here would be appreciated

cyoung, 2017-03-13:
Looks like your issue before was just startux continually opening `/dev/ttyAMA0` while you were using another program. Try shutting down the `stratux` service and seeing if the problem persists.

cyoung, 2017-04-19:
@johnpayne - reopen if that didn't fix it.

closed 2017-04-19 by cyoung

Austinthemighty, 2017-05-03:
how do you have the gps hooked up?