cyoung/stratux #543: error code when starting fancontrol.py

roperf15, 2016-12-11:
1. Stratux version:1.0r1 build dc62a3abba (9/21/2017)
2. Stratux config: pi 3

    SDR
      - [ ] single
      - [X ] dual

    GPS
      - [X ] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [X ] no

    power source: battery and shore power

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2) FF newest

    EFB platform: (e.g., iOS 9.2) IOS

    EFB hardware: (e.g., iPad Mini 2) IPAD 2

4. Description of your issue:

root@raspberrypi:/usr/bin# python fancontrol.py
When trying to run fancontrol.py from the command line I get the error below.  I went back to the original 1.0r1 image and ran fancontrol.py with no issues.  I installed selected updates sequentially and tested fancontrol.py on each one.  They all worked until I got to dc62a3abba (9/21/2017). I tested several of the newer builds as well and got the same error. The error seems to have been introduced after f477a8366e (9/5/2017).  That build tested good.  I did not test 31bbf1bc1b (9/19/21).  I'm a noob and I can't tell where the possible problem might be when looking at the code.  I can see that 11 sep is when fancontrol.py was daemonized.   Here is the received error.

Traceback (most recent call last):
  File "fancontrol.py", line 12, in <module>
    from daemon import runner
ImportError: No module named daemon

thx
Jim

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

johnpayne, 2016-12-14:
Jim,  You might be missing the python-daemon package?
You could try installing it:
`sudo apt-get install python-daemon`

cyoung, 2016-12-14:
@roperf15 - which base image?

roperf15, 2016-12-15:
@cyoung - I'm a noob and not sure what you are asking.  I started with the full base image of 1.0r1 downloaded from this site and then got the update files from the list of daily builds.  Does that answer your question?  Thx!

Jim

peepsnet, 2016-12-18:
https://github.com/cyoung/stratux/issues/493

low-and-slow, 2016-12-18:
I can't get the fancontrol.py script to run like I was able to with the stable version, however I don't need to. A lot has changed since the stable version, including which GPIO pin is called out in the code.  It was pin 11, but now it's pin 12. Give that a shot.

roperf15, 2016-12-21:
I did what johnpayne suggested above and it solved the problem.  Thanks for the help!

sudo apt-get install python-daemon

closed 2016-12-21 by roperf15