cyoung/stratux #545: Compile error : missing librtlsdr

johnpayne, 2016-12-14:
Just cloned the stratux repo.  Attempted to compile and got the following error:

root@JPi:/home/pi/stratux# make
make xdump978
make[1]: Entering directory '/home/pi/stratux'
cd dump978 && make lib
make[2]: Entering directory '/home/pi/stratux/dump978'
gcc -c -O2 -g -Wall -Werror -Ifec -fpic -DBUILD_LIB=1 dump978.c fec.c fec/decode_rs_char.c fec/init_rs_char.c
gcc -shared -lm -o ../libdump978.so dump978.o fec.o decode_rs_char.o init_rs_char.o 
make[2]: Leaving directory '/home/pi/stratux/dump978'
sudo cp -f ./libdump978.so /usr/lib/libdump978.so
make[1]: Leaving directory '/home/pi/stratux'
make xdump1090
make[1]: Entering directory '/home/pi/stratux'
git submodule update --init
cd dump1090 && make
make[2]: Entering directory '/home/pi/stratux/dump1090'
gcc -DMODES_DUMP1090_VERSION=\"v1.15-stratux-3-g9a4fb85\" -O2 -g -Wall -Werror -W `pkg-config --cflags librtlsdr`  -c dump1090.c -o dump1090.o
**Package librtlsdr was not found in the pkg-config search path.
Perhaps you should add the directory containing `librtlsdr.pc'
to the PKG_CONFIG_PATH environment variable
No package 'librtlsdr' found
dump1090.c:52:21: fatal error: rtl-sdr.h: No such file or directory
 #include <rtl-sdr.h>**
                     ^
compilation terminated.
Makefile:37: recipe for target 'dump1090.o' failed
make[2]: *** [dump1090.o] Error 1
make[2]: Leaving directory '/home/pi/stratux/dump1090'
Makefile:19: recipe for target 'xdump1090' failed
make[1]: *** [xdump1090] Error 2
make[1]: Leaving directory '/home/pi/stratux'
Makefile:10: recipe for target 'all' failed
make: *** [all] Error 2

Do I need to add this library somehow?

cyoung, 2016-12-14:
Yes, you need to compile and install librtlsdr. It's installed on the stock images (see releases). If you want to start from a clean image (raspbian, etc), then take a look at https://github.com/jpoirier/stratux-setup for guidance on the pre-requisites.

johnpayne, 2016-12-14:
That's that worked.  I have a RPi i'm trying to use a dev platform.  

I first needed to install libusb 1.0 with apt-get in order to compile librtlsdr.

now i'm hitting some sort of go error related to ipv4 and ipv6 (see below).  
Would it just be easier to try and compile on my stratux rpi itself?

Thanks

`# golang.org/x/net/ipv6
src/golang.org/x/net/ipv6/zsys_linux_arm64.go:7: sysIPV6_ADDRFORM redeclared in this block
	previous declaration at src/golang.org/x/net/ipv6/zsys_linux_arm.go:7
src/golang.org/x/net/ipv6/zsys_linux_arm64.go:8: sysIPV6_2292PKTINFO redeclared in this block
	previous declaration at src/golang.org/x/net/ipv6/zsys_linux_arm.go:8
src/golang.org/x/net/ipv6/zsys_linux_arm64.go:9: sysIPV6_2292HOPOPTS redeclared in this block
	previous declaration at src/golang.org/x/net/ipv6/zsys_linux_arm.go:9
src/golang.org/x/net/ipv6/zsys_linux_arm64.go:10: sysIPV6_2292DSTOPTS redeclared in this block
	previous declaration at src/golang.org/x/net/ipv6/zsys_linux_arm.go:10
src/golang.org/x/net/ipv6/zsys_linux_arm64.go:11: sysIPV6_2292RTHDR redeclared in this block
	previous declaration at src/golang.org/x/net/ipv6/zsys_linux_arm.go:11
src/golang.org/x/net/ipv6/zsys_linux_arm64.go:12: sysIPV6_2292PKTOPTIONS redeclared in this block
	previous declaration at src/golang.org/x/net/ipv6/zsys_linux_arm.go:12
src/golang.org/x/net/ipv6/zsys_linux_arm64.go:13: sysIPV6_CHECKSUM redeclared in this block
	previous declaration at src/golang.org/x/net/ipv6/zsys_linux_arm.go:13
src/golang.org/x/net/ipv6/zsys_linux_arm64.go:14: sysIPV6_2292HOPLIMIT redeclared in this block
	previous declaration at src/golang.org/x/net/ipv6/zsys_linux_arm.go:14
src/golang.org/x/net/ipv6/zsys_linux_arm64.go:15: sysIPV6_NEXTHOP redeclared in this block
	previous declaration at src/golang.org/x/net/ipv6/zsys_linux_arm.go:15
src/golang.org/x/net/ipv6/zsys_linux_arm64.go:16: sysIPV6_FLOWINFO redeclared in this block
	previous declaration at src/golang.org/x/net/ipv6/zsys_linux_arm.go:16
src/golang.org/x/net/ipv6/zsys_linux_arm64.go:16: too many errors
`

cyoung, 2016-12-15:
Maybe just use the v1.1b1 release image. All of the dev tools are there.

johnpayne, 2016-12-15:
Thanks, i'll try that.

johnpayne, 2016-12-21:
this worked perfectly.  I can compile and test now.

Thanks!

closed 2016-12-21 by johnpayne