cyoung/stratux #547: ADS OUT functionality with Stratux into a light plane

lcoulon, 2016-12-15:
Hello

Would you please tell me how I should connect Stratux to my Mode S transponder Funwerk TRT800 on the serial interface in order to get ADS-B Out functionality ?

I did not find any tutorial to do this.
Any help will be much appreciated.

Regards,

ssokol, 2016-12-15:
Stratux is an ADS-B Receiver (aka "ADS-B In"). It does not do ADS-B Out. You will need either a 1090-ES transponder or a UAT transmitter to add ADS-B Out.

closed 2016-12-15 by ssokol

AvSquirrel, 2016-12-15:
@ssokol - this doesn't appear to be the usual "can I use Stratux for ADS-B Out" question. The OP has an ES-capable transponder (Funkwerk TRT800), but lacks a position source.

@lcoulon - If you had an approved GPS position source, you would connect it to Pin 12 of the TRT800 (Pin 3 of the breakout cable). Input format would be NMEA, 4800 baud.

[image]

[image]

However, Stratux does not use an approved (e.g. TSO C145/146 or DO229 compliant) position source... those receivers have additional integrity checks and the GPS receivers we use in this project don't meet the position / velocity integrity reporting, failure detection, etc, requirements for use in ADS-B Out.

lcoulon, 2016-12-15:
Hello Keith,
Many thanks for the details it helps.
I want my light aircraft visible on ADS-b systems, even I fly with mode S enabled on the TRT-800 transponder , I've never been able to see my tracks visible on Flightradar24 logs. 
However control and towers always are able to receive my flight information such as ID, speed, altitude : all informations related to mode S
So I was thinking to add a GPS source output to my TRT800 transponder.
Does someone know if projects based from Raspberry and USB Glonass GPS exist and would do the job or I have to use a specific GPS output frame only issued from avionics GPS such as Garmin or other expensive kinds ?
Would you please tell me more about the purposes of optional USB GPS and AHARS daughter card used with Stratux ?
Regard,

cyoung, 2016-12-15:
@lcoulon - thanks for asking. This is interesting.

@AvSquirrel - which integrity checks are you seeing? I just see that it wants an RMC statement over RS-232 at 4800 baud.

http://fasvig.org/funke-support-for-uncertified-gps-ads-b-out

http://forum.pilotaware.com/index.php?topic=285.0 (scsirob's comments are particularly interesting).

@lcoulon - I assume you've done your research into the regulations aspect for wherever you are located. It seems like this transponder is used mostly outside of the US so I don't really know what applies to where you are. GPS NMEA output isn't something that stratux does currently but it could easily do so. Would you mind giving some background on why anyone else might want to do this so we can assess the value of adding it to the code?

lcoulon, 2016-12-16:
Hello

Adding ADS Out to Stratux will help lot of light plane pilots to get a fully working ADS-B system if they have Mode S transponder onboard.

At the moment most of light crafts are not properly detected by Flightradar 24 or Flightaware.

This probably because their flight levels do not permit to be seen by other larger planes very often.

I will really appreciate to test this new feature on my plane and provide you test results and feedback.

Regards,

AvSquirrel, 2016-12-16:
@cyoung , the NavWorx AD debacle has been on my mind, and I suppose I was begging the question, assuming that @lcoulon wanted to meet the performance requirements of a DO-260B / TSO-C166b compliant system. 

Appendix B of [AC 20-165B](http://www.faa.gov/documentLibrary/media/Advisory_Circular/AC_20-165B.pdf) outlines the FAA's guidelines for determining is a position source is suitable for use in an approved ADS-B system, and would assume that the EASA requirements (*if* an approved position source is required) would be similar. Among other parameters, this AC requires position, horizontal velocity, 95% probability horizontal and vertical accuracy, velocity accuracy, position integrity, and alerting of integrity faults (loss of RAIM, etc.)

If none of this matters, i.e. it's legal in EASA airspace (France?) to transmit ADS-B using an uncertified position source, and @lcoulon isn't trying to meet the equivalent of the FAA 2020 mandate) then I'd say there's no reason not to try. I've read through the PDF and it seems like the TRT800 just needs GPRMC messages at 1 Hz rate.  That should be dead simple to implement, and doesn't even require a Stratux; a receiver like the [BR-355-S4](http://usglobalsat.com/p-689-br-355-s4.aspx) should cost under €35, and provides the required messages. One would have to validate that the voltages are compatible with the TRT800 (and might need a 5V power supply for the GPS). If one had a Stratux with GPS and didn't want to buy a second GPS receiver, all you'd have to do is add a USB-RS232 adapter, and forward the GPRMC messages from the attached GPS to the serial port.

The only drawback to this is that SDA/SIL would be transmitted as "0" if this is a legal installation -- and that means that certified ADS-B In devices probably still won't be able to see you.

cyoung, 2016-12-16:
That's a good point - @lcoulon you could just hook this up directly:
http://www.usglobalsat.com/images/br355_optional_cable_sets.jpg

Is that something you could try ?

Curious to see how this works.

lcoulon, 2016-12-18:
Hello,

Actually i can't easily order that model as i'm in France.
However i ordered a VK162

It is  NMEA 0183 compliant , but i think a code should be added to Stratux to get it working.

If someone is able to add this functionnality to Stratux, i will be please to test it on my Mode S transponder.

cyoung, 2016-12-20:
@lcoulon - that's great, could you send me an email and we can coordinate a test? cyoung@gmail.com

lcoulon, 2016-12-20:
Hello

Once i receive my USB GPS VK162, we can do a test.

Do you have a part of code to add into Stratux ?

I also think we also should make some voltage levels translation in 
order to get the GPS signal output compatible with transponder ADS-b 
input requirements.

lcoulon, 2017-01-03:
Hello,

I received my USB GPS : VK162
Could we do some tried ?

Did someone implemented the GPS coordinates report in the code yet ?

lcoulon, 2017-01-13:
Hello

Could we no proceed some tests ?
I have received my USB GPS !