cyoung/stratux #552: description of issue

low-and-slow, 2016-12-18:
1. Stratux version: v1.1b1
2. Stratux config:

    SDR
      - [ ] single
      - [X ] dual

    GPS
      - [X ] yes
      - [ ] no
      type: vk162

    AHRS
      - [ ] yes
      - [ ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:

Hi, I wanted to check out the gpsattitude so I uploaded the appropriate sh file. However, I noticed when I did that, my fan wouldn't turn on (I'm running the fancontrol.py script at boot). I tried manually running the script in terminal and I'm getting:

 "File "fancontrol.py", line 12, in <module> from daemon import runner
ImportError: No module named daemon"

So I figured I'd try the beta update, and nothing changed. Any help is appreciated.

Thanks

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

peepsnet, 2016-12-18:
check your reddit post for info
https://www.reddit.com/r/stratux/comments/5j0njz/updated_to_v11b1/

peepsnet, 2016-12-18:
also 
https://github.com/cyoung/stratux/issues/543

low-and-slow, 2016-12-18:
Little did I know that the fancontrol.py script was already added to run at boot, so I wasn't expecting it to run at all when I first booted. I was expecting to have to add it myself to crontab. However, I was looking at the history of fancontrol.py and noticed the called out GPIO pin was changed from 11 to 12 for some reason. Once I swapped pins, the fan started right up.

Ergonomicmike, 2016-12-19:
So you can close this issue then?

closed 2016-12-19 by low-and-slow