cyoung/stratux #558: Unsigned 16-bit int is processed incorrectly in ForeFlight

ThatOneRoadie, 2016-12-28:
1. Stratux version: 1.0r1
2. Stratux config: Raspberry Pi 2 w/ Edimax

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [x] yes
      - [ ] no
      type: RY835ai (GPIO Unshielded)

    AHRS
      - [ ] yes
      - [x] no

    power source: 24000mAh Power Bank (5v 3A)

    usb cable: Anker MicroUSB 3 Feet

3. EFB app and version: (e.g., WingX Pro7 8.6.2) ForeFlight v7.6.1

    EFB platform: (e.g., iOS 9.2) iOS 9.0.2

    EFB hardware: (e.g., iPad Mini 2) iPhone 6s Plus

4. Description of your issue:

    Foreflight reports negative altitude after climbing past 32,767.5 feet. Example recorded track from ForeFlight during a descent from FL380: http://i.imgur.com/mlMRgl5.png indicating "minimum" altitude below 0 on the graph.

AvSquirrel, 2016-12-28:
See #447. Resolved in ForeFlight v7.7.

closed 2016-12-28 by ThatOneRoadie

Ergonomicmike, 2016-12-28:
Sorry, my fault for the duplicate issue. I encouraged Roadie to post his issue in the github before I searched.

AvSquirrel, 2016-12-29:
@Ergonomicmike and @ThatOneRoadie  - no worries at all! Sometimes a little tribal knowledge is easier / faster than knowing what to search for, and can give context or insight that you'd otherwise miss. 

One of those lessons is that development and testing on this project is usually happening against the latest released versions of EFB software, or at most, one version back. However, not all users upgrade right away. Bugs that we considered closed and forgotten about months ago may still be out in the wild, and new Stratux features don't always play well with old software.

Here's an example. ForeFlight 7.6's GDL90 message handling caused it to drop _all_ GDL90 messages -- that's traffic, weather, heartbeat... everything -- if it saw GDL90 AHRS messages.  (Remember back in March when we started telling users to turn off AHRS if they were using FF?) Within a couple weeks, Chris rolled out a temporary fix that disabled Stratux AHRS if ForeFlight was detected, and rolled that into the v0.9 images. ForeFlight fixed this on their end when they released ForeFlight v7.7 in July. Subsequent releases of 8.0, 8.1, and 8.2 have all worked fine.

Once FF 7.7 was in the wild, Chris took out the temporary fix before releasing v1.0. More recently (November 2016) he enabled all-the-time AHRS message output when the 'gpsattitude' branch was merged into the master branch.  The v1.1b1 image, and any builds newer than 6c50a96 therefore won't work with ForeFlight 7.6 unless you edit stratux.conf to remove `NETWORK_AHRS_GDL90` capability.