cyoung/stratux #561: Overlapping targets with dual antennae

dribble7, 2017-01-11:
1. Stratux version:1.1b1
2. Stratux config:

    SDR
      - [ ] single
      - [X ] dual

    GPS
      - [ ] yes
      - [X ] no
      type:

    AHRS
      - [ ] yes
      - [X ] no

    power source: battery
    usb cable: 12"

3. EFB app and version: Naviator 3.8.4
    EFB platform: Android 6.0.1
    EFB hardware: Nexus7v2013

4. Description of your issue:
I just upgraded to stratux v1.1b1, which I am running with v3.8.4 of Naviator.  I also upgraded from an RPi2 with a single, 978 Mhz antenna to an RPi3 with dual (978 & 1090) antennas.  Running this set-up close to KFFZ (very heavy traffic area), I can receive 1 ADS-B ground station.  I now notice that I am showing 2 slightly-overlapping targets for aircraft with ADS-B out, one from direct reception and one from the tower rebroadcast.
    I can tell that for the overlapping targets I see, the direct reception target is on 1090 MHz, because if I disconnect that antenna, the "ghost" target goes away.  Of course, the rebroadcast is always on 978 Mhz.  (There aren't enough 978 direct broadcast targets to see if they also would "ghost".)
   I might hope that either the stratux software or the Naviator software would notice the direct and rebroadcast targets with the same N-number, and not display the rebroadcast information, which of course is just a tad behind and makes the ghost. 
    So the questions is -  Is the issue with the stratux software or with Naviator?
Thanks
Andy Elliott 
N7v2, A6.0.1, Naviator 3.8.4, RPi3 w/stratux v1.1b1

Ergonomicmike, 2017-01-13:
I fly out of DVT and get the feed from the ADB tower at FFZ at pattern alt or higher.

I haven't noticed the ghosting that you're citing (I have dual band too). Flew last Sat. Plan to fly Friday and will look for it. (I have seen ocassional ghost targets NW of CGZ, where I'm probably receiving traffic from two disparate towers.)

My thinking is that the Stratux simply sends the data it receives to the EFB. 

I run iFly. You can doanload a free trial (their v9.8 should be out next week) and see if the problem goes away w/ a different EFB.

I'm Mode C. Are you Mode S?

Some of us are hoping that the FAA is starting to broadcast TIS-B in the blind, which would cause minor ghosting like this. But I've not seen any press about it if the FAA started doing that.

In any event, I don't think this is a bug w/ Stratux.

cyoung, 2017-01-13:
Can you see the ICAO identifier for the targets with Naviator? If not, look in the webui (http://192.168.10.1/).

Stratux de-duplicates targets based on the ICAO identifier - which should be equal for any given ADS-B out target, regardless of your data source (rebroadcast or direct).

dribble7, 2017-01-13:
Yes, in some cases both the targets have an N-number attached on the Naviator display.  The unit is in the aircraft now, but I'll get it out this weekend and do a couple of screen captures for you.

Ergonomicmike, 2017-01-14:
Oops - I stand corrected on Stratux sending raw data.
Flew Friday. No dupe targets in the Phx area.
That's with dual Stratux, the latest .sh update, latest iFly beta. (Which will be v9.8 real soon now.)

Ergonomicmike, 2017-01-14:
On de-dupe'ing: I presume that the Stratux removes the 978 duplicate, the 1090 being the more timely/accurate data? (Would make sense from a state diagram point of view, since 1090 data could come in first.)

cyoung, 2017-01-14:
The newest position report for any given ICAO address is used, regardless of the source of the data. If we knew that a 1090 target were in range and were also receiving its position on 978, you're probably right that it would be safe to discard the 978 report. Knowing whether the target is in range isn't so easy.

Ergonomicmike, 2017-01-14:
Okay. I assumed that 1090 would always come in first since it's direct xmission and not re-boadcast over 978. 

More critically, it seems that any 978 re-boadcast of a 1090 target will be stale data and so always discarded. 

I'm trying to understand how there could be a dupe if an aircraft with 1090 out wasn't received on 1090, but was received on 978. In the event of a "miss," where the 1090 xmission wasn't received one time, then there would only be the 978 data to forward to the EFB. (Hmmm... this is getting complicated fast. I presume there is a time window where the Stratux considers same ICAO target a dupe? (I'm assuming the Stratux does not look at position too for dupe rejection?))

In any event, it's academic. However it's doing it, it works.

Ergonomicmike, 2017-01-14:
Sorry to beat this to death, but what defines "first"? Say that a target is xmitting 1090, but the first time the Stratux sees it is on the TIS-B rebroadcast on 978. And say also that both are received in that order for the remainder of the time that that traffic is seen. Does that mean that the Stratux will always send the EFB the older, TIS-B 978 data?

dribble7, 2017-01-14:
OK, Since I definitely **am** seeing the dual targets, and since I am not an expert on stratux (just a user), I am going to the hangar and get the unit and upload a couple of screen save images.  Problem is that weather is poor here today and there may be few targets out there.  But once I do upload, is there some other data _cyoung_ will need for debugging?

Ergonomicmike, 2017-01-15:
I don't doubt that dribble7 is seeing dupe targets. If it's not his EFB (try a free trial of another EFB), I can fly over to FFZ someday to see if being next to a tower causes the problem on my setup. (Last time I was there, there was a restaurant on the field.)

I've flown past towers before w/o dupes. But maybe being right next to one causes a problem? (Although I'm grasping to understand how. (Race condition? I don't know where our main ATC antenna is here in the valley. I thought it was at KPHX. But perhaps it's at FFZ.) 

If proximity was a problem, wouldn't others similarly situated report the same problem?

Will be interesting to see what's going on.

Ergonomicmike, 2017-01-15:
Multipath? Reflections of the tower signal off hangars? (Does he get the dupes in the air too?)

dribble7, 2017-01-16:
Poor weather here again today, so didn't get the normal multitude of targets from the local flight school, but **did** manage to get 1 screen shot of the double target from an airliner passing into PHX.  See attached, where you will note that **both** targets have the ADS-B identifier.
[image]

dribble7, 2017-01-16:
Note that I **have** seen double targets with plain N-numbers attached, so it's not an issue of the call sign.  I will get another shot or two tomorrow when the weather clears, with doubled N-numbers.

Ergonomicmike, 2017-01-16:
Andy, if it's not your EFB that's the problem, is it possible that you have two SDR's that are set for 1090? (It's not a function of the antenna, but of the SDR itself.) The Stratux shouldn't let that happen, although I don't know if it tests for that condition. But two SDR's receiving 1090 might explain your dupe targets.

Airliners are not usually ADS-B out. So usually you wouldn't be in their puck to get a retransmission on 978.

An easy test - can you receive wx data (like NEXRAD or METARS) from the tower? 

Or - pull one SDR and look at the Stratux page in the Web UI showing on what freq the traffic was received. Then put the other back in and pull the first one out and tell us if traffic is still coming over 1090.

dribble7, 2017-01-16:
Yes, I definitiely have one SDR @978 and one @1090.  Both were marked and I have the corresponding antennas appropriately attached.  I **do** get FIS-B info, and when I remove the 1090 antenna, the ghost target disappears.  Based on the following by _cyoung_, I am _assuming for now_ that the problem is with the stratux: 
> Stratux de-duplicates targets based on the ICAO identifier - which should be equal for any given ADS-B out target, regardless of your data source (rebroadcast or direct).

(Yes, I know about "assume"!)  Once I upload a few more GA-type screenshots today, I hope that we'll get some more specific debugging requests from him, and maybe Mike from Naviator will chime in.

Ergonomicmike, 2017-01-16:
If you remove the SDR marked 978, do you still receive the tower? (The message count for 978 on the status page should start coming down immediately. It takes a minute for the tower count to drop to zero.) Maybe both SDRs are on 978? (Just trying to eliminate variables.)

It's so strange that you're always in someone's puck that you get a re-xmit of 1090 targets on 978. Do all tragets have dupes?

If the problem continues after you try another EFB, then we can make plans to meet at FFZ to see if my set up does the same thing as yours there. (Just to eliminate that this isn't a strange hardware, proximity or EFB issue.) FWIW, iFly has the ability to save raw ADS-B binary files. Don't know if that would help Chris.

Or, do you get dupes in the air or on the ground at DVT too?

dribble7, 2017-01-16:
> It's so strange that you're always in someone's puck that you get a re-xmit of 1090 targets on 978. Do all targets have dupes?

Not so strange, IMHO.  At home I am ~3 miles from the FFZ ADS-B tower and 200' higher. The CAE-PARC training facility at FFZ has >100 airplanes and every one is ADS-B out equipped.  I am directly underneath the departure path when FFZ is operating on 04.

dribble7, 2017-01-16:
This is a wide view showing a typical N-numbered ghost.  Note that the non-ADS-B-out aircraft (rebroadcast only) do not have a double target.
[image]

dribble7, 2017-01-16:
And here is a zoomed-in view of that same target. I think the ball is now in the stratux/Naviator court?

[image]

Ergonomicmike, 2017-01-16:
We won't know if it's Naviator (or something else) until you try another EFB and report on that.

Strange that other Naviator/Stratux users aren't reporting this. Do you know anyone there at FFZ with another brand of ADS-B receiver that you can connect to to try to see if the problem follows the Stratux?

Can you ask someone at the flight school if they get dupes?

If you're really in someone's puck (which it appears that you must be in order to get re-broadcasts), then do you ever get dupes for targets that are, say, 20 miles away? (i.e., outside the puck)

Maybe with all the school aircraft pinging the tower, this is what one gets in close proximity to an airport with an ADS-B tower?

If it is a Stratux problem, then perhaps the window time slice to check for dupes needs to be increased a bit more?

mmiller84, 2017-01-16:
@cyoung When you de-dupe the targets, what value is passed on to the EFB in the Address Type field? Naviator considers the unique ID for a traffic target to be the combination of {Address Type, Participant Address}, as per the GDL90 spec. If the address type is randomly alternated by the Stratux during the de-dupe process, Naviator would treat it as two separate targets and it could explain what @dribble7 is experiencing.

Thoughts?

dribble7, 2017-01-16:
Excellent catch Mike!  The GDL 90 spec says that within the Target Identity, the  _address type_ is **different** for a direct ADS-B target and a rebroadcast TIS-B target. 
> 3.5.1.2 TARGET IDENTITY
> The identity of a target is formed by the combination of the Address Type “t” along with the
> Participant Address "aaaaaa". Together these form a 28-bit field that uniquely identifies a given
> ADS-B or TIS-B participant. The Address Types are defined as follows:
> t = 0 : ADS-B with ICAO address
> t = 1 : ADS-B with Self-assigned address
> t = 2 : TIS-B with ICAO address
> t = 3 : TIS-B with track file ID.
> t = 4 : Surface Vehicle
> t = 5 : Ground Station Beacon
> t = 6-15 : reserved 

This puts us back in the position of asking **@cyoung** exactly what the stratux code does when he says
> Stratux de-duplicates targets based on the ICAO identifier - which should be equal for any given ADS-B out target, regardless of your data source (rebroadcast or direct).

"ICAO identifier" falls between "Target **Identity**" and "**ICAO** address"!  Does stratux actually de-duplicate on the same criteria as Naviator, based on the GDL90 spec definition of Target Identity, also requiring the Address Type to match?  That _would_ allow the ghosts to get out.  Or is some other algorithm used?

cyoung, 2017-01-16:
@mmiller84 - the target type is changed each time it is received according to the source the target was received from. So that would be the reason for the duplicated targets with Naviator's method of displaying. If we both strictly use GDL90's definition of "Target Identity", then any dual band system displaying duplicated targets would be expected behavior. I'm of the opinion that the ICAO address is a complete and sufficient identifier for a target, so that both sides should consider any target with equal ICAO addresses as equal. What do you think?

Ergonomicmike, 2017-01-16:
If it turns out that the Stratux totally squelches duplicate ICAO targets, then I wonder how often Naviator refreshes the target drawing?  Or if Naviator does any post-processing? I know that iFly refreshes traffic at  a lethargic 1 Hz. And iFly does some post-processing, in that targets will "coast" when lost. (IIRC, some of that coasting is in the Stratux too?) I think that iFly also does some filtering/smoothing of target data too, although I have seen dupe targets in flight a couple of times south of the Phx Class B, north of CGZ along the I-10 corridor. (Am not ADS-B Out, so maybe I was in someone's puck those times? Have done about 25 flights that way in the past year - seeing dupes there is the exception.)

Update: After seeing Chris's latest comment here, unless someone gathers the data first, I will plan to fly over to FFZ Friday to see if iFly and Avare have this dupe problem. (I see that Anzio's is no longer there. Any word on whether Stone & Hearth is worth the walk?)

kdknigga, 2017-01-16:
>  I will plan to fly over to FFZ Friday to see if iFly and Avare have this
dupe problem.

I've never noticed it with Avare.  I know that's hardly scientific.

Ergonomicmike, 2017-01-16:
Yeah, I can see that his is a somewhat unique case in that 1) he's very close to a tower and 2) he's always in a puck, because there are lots of school aircraft pinging the tower. (And lots of air carrier around too on 1090.)

When I land at KPAN, there's a tower about 10 miles away. I've not noticed any dupes there. But then there's not much traffic there anyway. 

When I fly over KPHX I'm receiving Andy's tower, but I might not be in anyone's puck. Or, as he points out, I might be zoomed out enough that the targets appear merged. (Although I've seen dupe targets at my normal zooms before. So I'm back to thinking that iFly is filtering in post.)

Only tie breaker that I can think of is to see what iFly and Avare do there. I know someone with a Stratux. I can ask to borrow that and try that too. (I will ask the iFly engineer how they handle this matter too.)

dribble7, 2017-01-17:
OK, so **@cyoung** and **@mmiller84** (stratux and Naviator developers) have identified the source of the ghosts, :)  

The following is my measly $.02:  The stratux box is a fairly general purpose ADS-B receiver. It might be that some not-specifically-EFB application would actually want to know the full Target Identity, include _both_ the Address Type _and_ the Participant Address. Therefore, _IMHO_ the stratux should continue to report the full Target Identity of most recent hit.

On the other hand, Naviator **is** a specifically-EFB application and so has no apparent use for the Address Type for purposes of in-flight traffic display.  So again _IMHO_, it is Naviator that should ignore that field in the Target Identity and only use the ICAO address portion to determine unique IDs for display purposes.

Footnote: As I understand it, under the current situation, the ghosting problem is **not** specific to 1090 direct/978 rebroadcast.  In fact, if there were 978 direct/978 rebroadcast it will also occur.

cyoung, 2017-01-17:
This is a display (EFB) issue. @dribble7 and @mmiller84 will have to come to some conclusion on whether or not displaying multiple targets for data from different sources is desired behavior or not. Please let us know what you guys settle on since if the behavior stays then we can expect more confusion.

Another ticket for the branch of the discussion on preferring data from one source over another: https://github.com/cyoung/stratux/issues/562

(only tangentially related to and NOT the cause of this issue).

cyoung, 2017-03-13:
@mmiller84 - did you make the change on your end? Closing it now since there is nothing further we can do, but if you can please update here so that it's documented. Thanks!

closed 2017-03-13 by cyoung

mmiller84, 2017-03-14:
@cyoung - yes, this change was implemented in the last release.