cyoung/stratux #564: Not recognized by Foreflight

captdmk, 2017-01-30:
1. Stratux version:
2. Stratux config:

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [X] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [X] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2). FOREFLIGHT

    EFB platform: (e.g., iOS 9.2). IOS 10.2.1

    EFB hardware: (e.g., iPad Mini 2). Ipad 2

4. Description of your issue:  I am a CFI and my student gifted me a Stratux for Christmas. I have 2 Ipads. My company ipad operating IOS 9.3.5 connects to Stratux. My personal ipad operating IOS 10.2.1 says No Internet Connection when I tap on wifi Stratux and Foreflight does not show it under the devices tab. Help!

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

johnpayne, 2017-01-30:
It is normal to see "no internet connection" as the wireless network doesn't connect you to the internet.  

Not sure why you are not seeing it in foreflight.  Maybe a foreflight guru will chime in here.  There could be a setting somewhere to enable ADSB?

jguluarte, 2017-03-11:
I would go near an airport with active traffic to test it.  You should just need to connect to the wifi and it should "just work"

captdmk, 2017-03-11:
I tried that.  Thanks anyway Justin.

> 
> I would go near an airport with active traffic to test it. You should just need to connect to the wifi and it should "just work"

David M. Klein
captdmk@comcast.net

cyoung, 2017-03-11:
@captdmk - do you see any Stratux information when you go to http://192.168.10.1/ in your browser (Safari or Chrome)?

What ForeFlight version are you running?

cyoung, 2017-04-19:
@captdmk - please reopen if you still have this issue. Not sure what happened, perhaps you have a very old version of ForeFlight. Have not heard of any other reports similar to this.

closed 2017-04-19 by cyoung

captdmk, 2017-04-19:
Still not working.  Have the latest version of Foreflight. 

David M. Klein
973-879-5740

cyoung, 2017-04-19:
Could you try another SD card? There have been a few instances where a user has one phone/tablet that connects, but another doesn't - and the issue is usually a physically damaged SD card.

Send me an email - cyoung@gmail.com. Thanks.

captdmk, 2017-04-19:
Replacing the SD card was the first thing we tried.   

David M. Klein
captdmk@comcast.net
973-879-5740

cyoung, 2017-04-20:
How many clients does the web interface show, http://192.168.10.1/?

captdmk, 2017-06-04:
> I am in the middle of an airline tour.  I will check when i get home on Saturday.  Thank you for your help!
>
> David M. Klein
> 973-879-5740
> Sent from my iPhone

I still cannot establish a connection between foreflight & the stratux.  Please help.   Here is some info:

Using Foreflight v9.1
Apple Ipad 2 - IOS 10.3.1
Wifi - connects to Statux
Here are some screen shots.   

David M. Klein
captdmk@comcast.net
973-879-5740

yeo423, 2017-08-04:
I seem to be having the same issue, was this ever resolved?

captdmk, 2017-08-04:
No.  It was never resolved.   Very disappointed.  

David M. Klein
973-879-5740

cyoung, 2017-08-04:
Didn't ever get an email from you. I'd send you an SD card to try out.

captdmk, 2017-08-04:
I sent a bunch of emails.  I guess you never received them.  I am at my house so I will not receive any data.  But It does find traffic but it will not display on my Foreflight.     Here are a bunch of screen shots from today.  Please help.  

David M. Klein
captdmk@comcast.net
973-879-5740

cyoung, 2017-08-18:
Issue was resolved - SD card or image writing issue.