cyoung/stratux #567: Suggestion: Blinking green light

Nokomis449, 2017-02-16:
Stratux issues can be boiled down to 50% power supply issues, 50% out-of-disk-space issues, and 50% Operator Headgap issues.  Since the green light is now used to signify that Stratux is up and running, I suggest that it be made to blink steadily when disk space is running low.

JohnOCFII, 2017-02-16:
I'd rather see the slow blink for Operator Headgap issues, but I know that'll be a b*tch to code. :)

youcangetme, 2017-02-17:
I would personally prefer leaving the green led showing network activity as it is today. That is step one in debugging an unresponsive system. I would broadcast via UDP the warnings in the GDL90 protocol to signal issues like disk space being low. The user then would know the unit needs to be checked via SSH for more information. Any warning via the GDL90 protocol will be available for viewing in any of the top 4 apps on Android (DroidEFB, Naviator, iFly GPS and Avare).

JohnOCFII, 2017-02-17:
> I would personally prefer leaving the green led showing network activity as it is today.

It no longer functions that way in the current release. That changed with the current release, v1.2r1.
The light goes to solid green, and stays that way once Stratux software is operational.

youcangetme, 2017-02-17:
OK, so basically the green light means nothing. Since it's not an open drain Stratus does not need to renew the led light on and thus it can crash and still appear "operational."

cyoung, 2017-02-17:
@youcangetme - the green light never blinked on network activity, it blinked previously on SD card write activity. That is the default behavior. I think the way it is now (turns on when Stratux starts up successfully) communicates more information than it did previously. Disk writes happen even without Stratux running. I'm not sure if it blinks on failed SD card write attempts or not.

A lot of support requests come from EFB issues from users just getting set up. Having the green light indicating that Stratux started up successfully has been helpful.

@Nokomis449 - good idea, we could make it slow blink whenever a "critical" system happens like an SD card issue. Oh, on the latest release the partitions are auto expanded on first boot.

youcangetme, 2017-02-17:
So you are saying is Stratux does not notify the client applications that it is a state where system requires maintenance?

Nokomis449, 2017-02-18:
@cyoung You can mark this as an enhancement if you'd prefer.

TravGoneFlying, 2017-03-04:
If I'm reading this right, the LED is repeatedly turned on once per second in heartBeatSender().

You could toggle it once per second so that it flashed the heartbeat at 1/2 Hz.

cyoung, 2017-04-19:
@youcangetme - good idea on the "maintenance required" bit. This is now implemented.

cyoung, 2017-04-19:
@Nokomis449 - added it.

closed 2017-04-19 by cyoung