cyoung/stratux #568: Thread-unsafe handling of pingResponse in network.go

AvSquirrel, 2017-02-20:
1. Stratux version: 1.2r1 + commmits (0ee94ff925a2d5)
2. Stratux config:

    SDR
      - [ ] single
      - [x] dual

    GPS
      - [x] yes
      - [ ] no
      type: RY725AI over USB (equivalent to VK-162)

    AHRS
      - [ ] yes
      - [x] no

    power source: Wall-wart adapter

3. EFB app and version: ForeFlight 8.3 (sleeping on two devices)
    EFB platform: iOS 10.2.1
    EFB hardware: iPad Air 2 and iPhone 6 (both devices sleeping)

4. Description of your issue:

I was running a long term test on commit 0ee94ff925 to evaluate UAT TIS-B suppression. A little over two days into the run, gen_gdl90 crashed. The stack trace indicated that there is an unsafe handling of the `pingResponse` map in network.go

```
fatal error: concurrent map read and map write

goroutine 49 [running]:
runtime.throw(0x4d7f57, 0x21)
        /root/go/src/runtime/panic.go:566 +0x78 fp=0x10a9eb0c sp=0x10a9eb00
runtime.mapaccess2_faststr(0x47c578, 0x109e2820, 0x109f9800, 0xd, 0x13152d80, 0x2)
        /root/go/src/runtime/hashmap_fast.go:306 +0x64 fp=0x10a9eb3c sp=0x10a9eb0c
main.isSleeping(0x109f9800, 0x12, 0x10a9ec30)
        /root/stratux-dev/main/network.go:151 +0xa4 fp=0x10a9ebcc sp=0x10a9eb3c
main.sendToAllConnectedClients(0x1427f3a0, 0x14, 0x20, 0x4, 0x3d4d6, 0x0, 0x1c9c380, 0x0)
        /root/stratux-dev/main/network.go:176 +0x148 fp=0x10a9ed44 sp=0x10a9ebcc
main.messageQueueSender()
        /root/stratux-dev/main/network.go:367 +0x10c fp=0x10a9efd4 sp=0x10a9ed44
runtime.goexit()
        /root/go/src/runtime/asm_arm.s:998 +0x4 fp=0x10a9efd4 sp=0x10a9efd4
created by main.initNetwork
        /root/stratux-dev/main/network.go:644 +0x1a8
```

The network code in my test build is the same as 1.2r1, and is unchanged since 6198fc2  (12/25/2016). [`isSleeping()`](https://github.com/cyoung/stratux/blob/6198fc2ce34f8b230066d42d21672784ef19cd8d/main/network.go#L148-L158) contained the read of `pingResponse` that triggered this crash; there also appears to be an unprotected read in [`getNetworkStats()`](https://github.com/cyoung/stratux/blob/6198fc2ce34f8b230066d42d21672784ef19cd8d/main/network.go#L304) and an unprotected write in [`sleepMonitor()`](https://github.com/cyoung/stratux/blob/6198fc2ce34f8b230066d42d21672784ef19cd8d/main/network.go#L524).

```
func isSleeping(k string) bool {
	ipAndPort := strings.Split(k, ":")
	// No ping response. Assume disconnected/sleeping device.
	if lastPing, ok := pingResponse[ipAndPort[0]]; !ok || stratuxClock.Since(lastPing) > (10*time.Second) {
		return true
	}
	if stratuxClock.Since(outSockets[k].LastUnreachable) < (5 * time.Second) {
		return true
	}
	return false
}
```

Full stack trace:
```
fatal error: concurrent map read and map write

goroutine 49 [running]:
runtime.throw(0x4d7f57, 0x21)
        /root/go/src/runtime/panic.go:566 +0x78 fp=0x10a9eb0c sp=0x10a9eb00
runtime.mapaccess2_faststr(0x47c578, 0x109e2820, 0x109f9800, 0xd, 0x13152d80, 0x2)
[... 377 lines cut ...]
bufio.(*Reader).ReadLine(0x10bdce40, 0x0, 0x0, 0x0, 0x64fc00, 0x0, 0x0)
        /root/go/src/bufio/bufio.go:359 +0x60
net/textproto.(*Reader).readLineSlice(0x10a98020, 0x0, 0x0, 0x0, 0x0, 0x0)
        /root/go/src/net/textproto/reader.go:55 +0x58
net/textproto.(*Reader).ReadLine(0x10a98020, 0x0, 0x0, 0x0, 0x0)
        /root/go/src/net/textproto/reader.go:36 +0x34
net/http.readRequest(0x10bdce40, 0x0, 0x11e99200, 0x0, 0x0)
        /root/go/src/net/http/request.go:793 +0x80
net/http.(*conn).readRequest(0x11187b80, 0x6553a0, 0x1333e640, 0x0, 0x0, 0x0)
        /root/go/src/net/http/server.go:765 +0x3d4
net/http.(*conn).serve(0x11187b80, 0x6553a0, 0x1333e640)
        /root/go/src/net/http/server.go:1532 +0x910
created by net/http.(*Server).Serve
        /root/go/src/net/http/server.go:2293 +0x470

goroutine 1311 [runnable, locked to thread]:
github.com/jpoirier/gortlsdr._Cfunc_rtlsdr_read_sync(0x73f016e8, 0x116a6000, 0x40000, 0x110464bc, 0x0)
        github.com/jpoirier/gortlsdr/_obj/_cgo_gotypes.go:382 +0x3c
github.com/jpoirier/gortlsdr.(*Context).ReadSync(0x73f016e8, 0x116a6000, 0x40000, 0x40000, 0x40000, 0x40000, 0x0, 0x0)
        /root/go_path/src/github.com/jpoirier/gortlsdr/rtlsdr.go:550 +0x18c
main.(*UAT).read(0x10c140a0)
        /root/stratux-dev/main/sdr.go:148 +0x200
created by main.createUATDev
        /root/stratux-dev/main/sdr.go:404 +0x228
```

cyoung, 2017-02-20:
Good find.

westphae, 2017-02-21:
I've been encountering this panic also.  It occurs regularly when gen_gdl90 is built with Go versions 1.6 or higher.  Runtime detection of concurrent map reads/writes was introduced in Go 1.6:  see https://golang.org/doc/go1.6#runtime.  @AvSquirrel , were you compiling with a more current version of Go?

pingResponse isn't the only map that suffers this panic.  I protected pingResponse with a mutex in my fork and subsequently also had a panic due to concurrent read/write of outSockets at https://github.com/cyoung/stratux/blob/master/main/network.go#L472.  My fork implements AHRS and runs at a higher CPU load, which I believe tends to make these panics occur more frequently.

Ergonomicmike, 2017-02-21:
Regarding CPU load under AHRS: Will it be necessary to change the governor for AHRS? Last I knew, the Stratux was throttled back quite a bit, but loafing along with normal ADS-B loads. (Hmmm... Maybe this is why sqlite logging lags?)

westphae, 2017-02-21:
I don't think it will be necessary to increase clock speed for AHRS.  We want the battery to last as long as possible!  I have the algorithm set to update more often than necessary while testing, so it's at a higher load now than it really needs to be.

But I suspect that the higher load increases the rate of thread collisions for this issue.  I had tried to reproduce this panic a few weeks ago without AHRS (so cyoung/stratux code but built with Go 1.7.5) and didn't get a panic after running for 24+ hours.  With my fork (no significant changes to network.go) it can happen within an hour without the mutex, even if the load is only 1.7 or so.

Mutex protection seems to resolve this issue for the pingResponse map, but before submitting a pull request I wanted to protect outSockets also.

westphae, 2017-02-21:
Looking through network.go, netMutex already exists to protect outSockets.  netMutex seems mainly intended to protect outSockets from simultaneous writes (which also happens to protect some of the reads).  In two places where outSockets is only read from, it is not protected by netMutex, namely https://github.com/cyoung/stratux/blob/master/main/network.go#L291 and https://github.com/cyoung/stratux/blob/master/main/network.go#L472.  The latter is the one that usually gives me grief).

How about if we change netMutex to a sync.RWMutex and protect the other reads with RLock()?  pingResponse also needs a sync.RWMutex.  Potentially it could be covered by netMutex--it is only written to at https://github.com/cyoung/stratux/blob/master/main/network.go#L524--but possibly it would lock for too long?  I set up a separate Mutex and in my limited test case that seems to do the trick so far.

cyoung, 2017-03-03:
@westphae - just skimmed your branch, on increasing load causing collisions, I don't think it relates to system/CPU load so much as it does an increased packet rate (and more frequent calls to `sendToAllConnectedClients()`. Increased number of clients connected, packet rate, and decreasing the traffic packet maximum size (https://github.com/cyoung/stratux/commit/3e663c89c8a812cb5f3f482f0c933bc46152c842) were all probably contributing factors.

cyoung, 2017-03-03:
Since most of the time that `netMutex` should be locked is within `messageQueueSender()`, and all of the other events which lock it should not occur often - using `netMutex` to protect all three of the network maps seems like the simplest approach. https://github.com/cyoung/stratux/commit/0360e506ec3336be77006c5ec3c4d2b3b3a860fc

* `icmpEchoSender()`
* `sleepMonitor()`
* `getNetworkStats()`
* `handleClientsGetRequest()`

This isn't the ideal approach, since it allows the network output stream to be affected by external forces (ICMP packets, HTTP requests) - so it would be easy to construct a DoS attack on the system.
Looking at the code again, it seems we had this problem all along. So this doesn't fix the DoS vulnerability issue, but should fix the normal operations issue. Let's try this approach for now and revisit the network code some time soon in the future, since there is a lot that can be simplified here.

cyoung, 2017-03-03:
http://updates.stratux.me/builds/update-stratux-v1.2r1-0360e506ec.sh

AvSquirrel, 2017-03-04:
@cyoung - I'll merge 0360e50 into my TIS-B suppression branch and give it a shot. (Haven't had any time to look at it since my last commit, but maybe I'll poke around this week.) Thanks!

@westphae - I had been compiling with go1.7.2 linux/arm, and panics were repeatable, taking anywhere from two hours to three days to bring down gen_gdl90. However, I see in the [go release notes](https://golang.org/doc/devel/release.html) that:

>go1.7.2 should not be used. It was tagged but not fully released. The release was deferred due to a last minute bug report.

and that [go1.8](https://golang.org/doc/go1.8) released a couple weeks ago, with an improved compiler and a more responsive garbage collector (i.e. fewer / shorter pauses). I'm going to install and give it a whirl.

AvSquirrel, 2017-03-04:
Huh. That was unexpected.

On go1.8 on 0ee94ff92 (old network code) panicked after 24 seconds, but on a different map variable ([`outSockets`](https://github.com/cyoung/stratux/blob/6198fc2ce34f8b230066d42d21672784ef19cd8d/main/network.go#L472)).

```
fatal error: concurrent map iteration and map write

goroutine 74 [running]:
runtime.throw(0x467ae5, 0x26)
        /root/go/src/runtime/panic.go:596 +0x70 fp=0x10a14630 sp=0x10a14624
runtime.mapiternext(0x10a14760)
        /root/go/src/runtime/hashmap.go:737 +0x6cc fp=0x10a1467c sp=0x10a14630
main.icmpEchoSender(0x10a6c0a0)
        /root/stratux-dev/main/network.go:472 +0x144 fp=0x10a147e4 sp=0x10a1467c
runtime.goexit()
        /root/go/src/runtime/asm_arm.s:1017 +0x4 fp=0x10a147e4 sp=0x10a147e4
created by main.sleepMonitor
        /root/stratux-dev/main/network.go:506 +0x68
```

I think I'll revert to go1.7.5 for now.

cyoung, 2017-03-04:
"concurrent map iteration and map write", that's actually another issue that needs to be fixed.

AvSquirrel, 2017-03-04:
@cyoung, FWIW I reinstalled go1.8 and compiled with the mutex-protected network code from 0360e50. 

Ran overnight (10 hours) with no obvious issues. I'll continue bench testing and hope to take if flying this weekend.

westphae, 2017-03-07:
@AvSquirrel I had the panic on outSockets also.  I don't remember having a panic on simultaneous iteration and write, but I guess that's just a variation on reading and writing.  I've had the same results whether using 1.7.5 or 1.8.

@cyoung I thought maybe the increased load caused everything to slow down, increasing the likelihood of a collision, but I could be wrong.  I had the panics even with only one client connected, and I don't think the AHRS is sending any more data.  In any case, protecting all the reads/writes with a mutex solved it for me.  I ran for a few days straight without any more panics, as well as over a few dozens of hours separately on quite a few test flights and bench runs.  I only protected outSockets and pingResponse, and decided that one mutex per map (i.e. just one additional mutex, though I made it a RWMutex instead of a plain Mutex) wasn't overly complicated and should minimize unnecessary blocking.  So far it's been trouble-free.  You said there's a third map you protected?  If your solution works, I'll change to your approach next time I merge your code into mine just to minimize differences.

westphae, 2017-03-07:
Ah, I see it: dhcpLeases is the third map.  I guess that one can't be written to and read from simultaneously anyway.

cyoung, 2017-03-07:
That's it - was not necessary but it might be read from in web calls in the future.

cyoung, 2017-03-30:
Testing seems good, changed as above and released in v1.2r2.

closed 2017-03-30 by cyoung