cyoung/stratux #569: Consistant card corruption on power loss

pdelbarba, 2017-02-25:
1. Stratux version:
2. Stratux config:

    SDR
      - [ ] single
      - [x ] dual

    GPS
      - [x ] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [ ] no

    power source: 2A

    usb cable: Anker

3. EFB app and version: Foreflight

    EFB platform: iOS

    EFB hardware: iPad Mini 4

4. Description of your issue:

On first power up, everything is great and works as expected.  If you unplug the unit without going into the web interface and selecting "shutdown" however, the SD card becomes corrupted and the stratux application will not open the access point.  Did not have this issue in 1.0 for whatever reason but it's been a significant issue in 1,2.

cyoung, 2017-02-25:
Do you have errors on the console/fsck output? Is the only symptom you're seeing that the access point does not come up? What SD card type, speed, brand, and source?

pdelbarba, 2017-02-27:
I'll do some testing when I get a chance.  Currently in the middle of IFR training so it's hard find time to have my system down.  

I can tell you that I've tried both SanDisk (32GB UHS-1) and Lexar (32GB UHS-1) cards and both failed the same way.  Both were purchased through amazon and have not had issues with other devices.

Currently the only symptom I'm seeing is a failure to bring up the access point.  I connected to a display and was able to log in and poke around without issue.  Systemd didn't appear to report any issues on startup but I didn't do a thorough examination of dmesg or any other logs.

crablab, 2017-03-29:
I'm having this issue as well. Card isn't "corrupted" persay but no AP is started. 
I'm using a Pi Zero W.

cyoung, 2017-03-29:
Different issue - the default image isn't configured for the Pi Zero W.

crablab, 2017-03-29:
@cyoung Okay, well I opened #579 anyhow

cyoung, 2017-04-19:
@pdelbarba - try v1.2r2 and reopen if you have the issue again.

closed 2017-04-19 by cyoung