cyoung/stratux #570: description of issue

ict-flying, 2017-03-02:
Stratux version: 1.2r1

    Stratux config: 9ee4617

    SDR
        single
        [X ] dual

    GPS
        [X ] yes
        no
        type: VK162

    AHRS
        [ X] yes
        no

    power source: Recommended Battery Pack - full charge

    usb cable:

    EFB app and version: (e.g., WingX Pro7 8.6.2) ForeFlight

    EFB platform: (e.g., iOS 9.2) Latest

    EFB hardware: (e.g., iPad Mini 2) iPad Pro 64G

    Description of your issue: UAT towers are always at 0 -> even though I know I am close to one.     Weather never gets updated?

c150heavyn, 2017-03-03:
I've had this issue when I had a failed SDR.   Have you tried configuring your other SDR (assuming it's working OK on 1080) as a 978 radio?

TravGoneFlying, 2017-03-04:
Towers have restricted downward signal. I can get good signal from 3rd floor parking garage but not from the ground level even though I'm line of sight in both situations.

On the other hand, atmosphere makes a difference as radio operators know. Some days, I can get good signal in my living room which is not line of sight. Some days zip.

In the air, of course, no problem.

cyoung, 2017-03-13:
@ict-flying - try going flying and see if you can pick up any towers in the air. You can also verify tower locations using Google Earth with this file:

https://dl.dropboxusercontent.com/u/58910037/user_waypoints.kml

Please re-open the ticket if you have the issue even in the air. Thanks!

closed 2017-03-13 by cyoung