cyoung/stratux #571: FLARM NMEA over USB -> RS232 for flight computers unable to integrate via UDP or TCP

soarak, 2017-03-05:
There are many soaring computers which are not able to take advantage of UDP and TCP protocols. These computers integrate via RS232 e.g. LX9000 series, ClearNav series, Zeus and many more. UDP and TCP are mostly supported by portable devices. Traditional glide computes do not support UDP and TCP. The RS232 has been a standard for these computers for many years. There are thousands of these soaring computers around the world.

Would you mind supporting USB->RS232 again. I think, the original support for FLARM was started by @AvSquirrel and it was done through USB->RS232 however for some reason the protocol was changed. Please reconsider porting this feature to USB->RS232 as well. This would be most appreciated.

jmcarlyle, 2017-03-05:
A few of us have been experimenting with v0.8r, which uses the RS232 Flarm output. We've had success in getting 978 UAT targets displayed on our Flarm displays with this version of code. But, we've also identified a number of minor issues that really should be addressed to make this variant more useful and reliable for the general user. I'd like to recommend that no changes be made to the code until we've had a chance to post our findings to this board (we're still gathering input from the group). Thanks!

-John

JS1CG, 2017-03-05:
I concur with the postings above, and definitely appreciate the assistance of those that have pioneered this project. I've had success with AVSquirrel's update (v0.8r) and UDP/TCP (v0.8r2) loaded onto the Straux base image (v1.0r1) ported to XCSoar. However, some of the more common soaring EFIS devices carried in gliders (Naviter Oudie, LXNAV FlarmView and LX90x0) will not display traffic with the most recent updates. They will, however, if the NMEA sentences $GPGGA and $PFLAU from a Power FLARM device are multiplexed with the $GPRMC and $PFLAA from the Stratux. I'd be interested in seeing how the Stratux works with all four NMEA strings ported to the GPIO (with TTL to RS232 level convertor), as this could allow the equipment to get back into the tiny Stratux box, and make it viable for cramped glider cockpits. Thanks for considering!
Charlie

ASW27B, 2017-03-06:
I started this project request about a year ago. Thanks @cyoung and @AvSquirrel for supporting it, and @ssokol for recommending it.

The UDP/TCP request basically sidetracked the project's usefulness for the original purpose of making Stratux available to the glider community - for the reasons described above. We have been using  v0.8r1 ever since with some bugs and problems associated with using USB-serial converters. The original vision was to ultimately move to serial output directly off the RPi, and eliminate the USB-Serial converter. This is highly desirable for a number of reasons we can enumerate if helpful.

The practical reality of ADS-B Out for gliders has raised a lot of interest in the goals of this project, including interest from the two of the most widely deployed glider computer manufacturers - LXNav and ClearNav.

The four people posting on this thread have gotten together to work out the issues that have been lingering since the last release. We've got several systems up and running with v0.8 and should be ready to do testing in a week or two. We are also working on a handful of (seemingly minor) bug fixes and enhancements. Serial output is probably the biggest ask.

We'd love some support to get this project back into working order. Thanks!

cyoung, 2017-03-07:
Could someone put together a collection of protocol documentation, requested hardware/platform support (with justification), and any development tools like simulators or free apps that can be used to work on this?

jmcarlyle, 2017-03-07:
Sure, we (the four people posting above your message) will be glad include such information as part of our findings. We are currently cooperatively working out a consensus report, with the goal of having everything in one place rather than scattered among various messages. This should make it easier for you to consider our requests. We appreciate your input.

AvSquirrel, 2017-03-08:
@soarak et al:

I reimplemented my UDP / TCP based work from last September on top of the latest commits to v1.2, and added basic serial out support for two types of USB adapters.

Supported outputs

- Port 10110 UDP
- Port 2000 TCP (AIR Connect format; )
- USB-to-serial output on CP2102 bridges (`/dev/serialout0`) and optional USB-to-RS232 output on Prolific 2303 bridges (`/dev/prolific0`)

Note that Prolific support will disable any GPS that uses that bridge, so this won't be compatible with a BU-353-S4 (you'll have to use one of the U-blox devices -- VK-162, etc.)

Traffic output was verified by displaying it on the SoarXC radar window, by connecting a terminal to port 2000, and by connecting a terminal to the RS232 output of a Prolific-based adapter. There are probably a few features

See https://github.com/AvSquirrel/stratux/tree/flarm-network-v12 for latest code. Commit 1b17cb4 is the most recent at the time of this post, and an update .sh is attached.

[update-stratux-flarm-v1.2r1-1b17cb4101.sh.zip](https://github.com/cyoung/stratux/files/826743/update-stratux-flarm-v1.2r1-1b17cb4101.sh.zip)

For what it's worth, I probably won't do much more development on this branch -- I lack the glider computer hardware and/or Euro-style EFB subscriptions to really be effective with this, nor do I have the background to understand how you expect targets to alert. However, time permitting, I'll try to answer any questions you have if you want to take this code and build on it.

AvSquirrel, 2017-03-08:
@JS1CG 

> However, some of the more common soaring EFIS devices carried in gliders (Naviter Oudie, LXNAV FlarmView and LX90x0) will not display traffic with the most recent updates. They will, however, if the [...]

This is actual output of 1b17cb4, recompiled with demo traffic enabled, piped out of my RPi with [this Prolific-based adapter](https://www.amzn.com/B0007T27H8/) and received on an old laptop with a DE-9 RS232 port. Each second, three GPS-related NMEA sentences are followed by as many PFLAA sentences are needed to describe nearby traffic, followed by one PFLAU sentence.

```
$GPRMC,054030.20,A,4453.13538,N,09324.88220,W,0.0,0.0,080317,,,D*42
$GPGGA,054030.20,4453.13538,N,09324.88220,W,2,8,1.00,262.7,M,-30.7,M,,*5A
$GPGSA,A,3,,,,,,,,,,,,,1.0,1.0,1.0*33
$PFLAA,0,12577,-2226,3283,1,FF0003!DEMO3,79,,101,0.0,8*58
$PFLAA,0,2824,-2400,609,1,FF0000!DEMO0,35,,25,0.0,8*6A
$PFLAA,0,21146,-18008,5067,1,FF0005!DEMO5,48,,235,0.0,8*63
$PFLAA,0,18326,10950,5958,1,FF0006!DEMO6,122,,177,0.0,8*71
$PFLAA,0,8424,191,7741,1,FF0008!DEMO8,91,,63,0.0,8*48
$PFLAA,0,-19223,13264,1581,1,48414C!SCX426,68,,148,-10.7,0*4F
$PFLAA,0,20776,-11107,9524,1,FF000A!DEMO10,60,,197,0.0,8*2F
$PFLAA,0,12458,11334,10416,1,FF000B!DEMO11,135,,139,0.0,8*01
$PFLAA,0,24393,8457,1500,1,FF0001!DEMO1,110,,216,0.0,8*43
$PFLAU,9,1,2,1,1,,0,,*68
$GPRMC,054031.10,A,4453.13515,N,09324.88220,W,0.0,0.0,080317,,,D*4F
$GPGGA,054031.10,4453.13515,N,09324.88220,W,2,9,1.00,262.6,M,-30.7,M,,*57
$GPGSA,A,3,,,,,,,,,,,,,1.0,1.0,1.0*33
$PFLAA,0,12596,-2126,3284,1,FF0003!DEMO3,79,,101,0.0,8*53
$PFLAA,0,2845,-2385,609,1,FF0000!DEMO0,36,,25,0.0,8*64
$PFLAA,0,21302,-17833,5067,1,FF0005!DEMO5,48,,235,0.0,8*6E
$PFLAA,0,18229,11100,5958,1,FF0006!DEMO6,123,,177,0.0,8*72
$PFLAA,0,8422,255,7741,1,FF0008!DEMO8,92,,63,0.0,8*46
$PFLAA,0,-19223,13264,1558,1,48414C!SCX426,68,,148,-10.7,0*4B
$PFLAA,0,20872,-10935,9524,1,FF000A!DEMO10,61,,197,0.0,8*2D
$PFLAA,0,12358,11433,10416,1,FF000B!DEMO11,135,,0,0.0,8*0D
$PFLAA,0,24319,8660,1500,1,FF0001!DEMO1,110,,216,0.0,8*47
$PFLAU,9,1,2,1,1,,0,,*68
```

ASW27B, 2017-03-08:
Thanks Keith,

If it would help, I might be able to get you a couple of the more common glider display devices. 

We'll get this version running and see what questions come up. One thing that probably will be an issue is the <AlarmLevel> parameter in the $PFLAU sentence, which shows set to Level 1 in you example output. Do you remember how you determined how to set this collision alarm?  

One other general question: Does Stratux de-duplicate targets, for instance the same target received via ADS-B direct and ADS-R, or is that done by the display applications (or not at all)?

Regards,

Andy

cyoung, 2017-03-13:
Move to #489.

closed 2017-03-13 by cyoung