cyoung/stratux #572: Stratux with Air Navigation Pro

ecsaulnier, 2017-03-07:
1. Stratux version: 1.2r1
2. Stratux config:

    SDR
      - [ ] single
      - [X ] dual

    GPS
      - [X ] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [X ] no

    power source: Battery

    usb cable: 

3. EFB app and version: Air Navigation Pro last version

    EFB platform: IOS 10.2.1
    EFB hardware: iPhone 6 +

4. Description of your issue:
I want use Air Navigation Pro with Stratux ads-b but this app ask me :

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

cyoung, 2017-03-07:
You have the IP and port correct. There is no passcode. Is it working without?

ecsaulnier, 2017-03-07:
Not that does not go. I very tested. The raspberry walk well and via the address 192.168.10.1 one recovers the traffic well.

That goes with the applications WingX Pro7 and AvPlan EFB.

On Air Navigation Pro with this param :

192.168.10.1, Port 4000 and NULL or 0 or 6000 in the Pin Code

it doesnt walk

AvSquirrel, 2017-03-08:
@ecsaulnier -

"Passcode" implies that Air Navigation Pro uses the FLARM network format, which is used by devices such as AIR Connect. This is different that the GDL90 network format used by most American EFB applications (and Stratux).

I developed a branch of the Stratux software that emulates the FLARM  output, but it hasn't been tested extensively. If you'd like to try it, [read this comment](https://github.com/cyoung/stratux/issues/571#issuecomment-284955097), then download and install the linked .sh update file.

Once it reboots, you should be able to connect to Port **2**000, and use any value in the PIN code.

ecsaulnier, 2017-03-08:
Thank you for your answer.

I want to test your new version well

Do you have a procedure to install this new version on the raspberry?

ecsaulnier, 2017-03-14:
I thank you for all this help.

I succeeded in installing the update 1b17cb4 and now on Air Navigation Pro with the TCP port the 2000 led concerning the traffic passes well to the green.

A new question: This is the real traffic which passes by the port 2000 or the traffic DEMO because it has there nothing which appears on the cartography Air Nav Pro

Thank you in advance for all that you do.
Pilot French Autogiro

ecsaulnier, 2017-04-03:
I have just installed the version 1.2r2-dev version, can-you indicate to me how to put over the version which manages protocol FLARM on the port 2000 for the application Air VOR Pro. 
Thank for your response

French Pilot.

closed 2018-01-17 by cyoung