cyoung/stratux #573: Stratux making more EMI than it used to

Ergonomicmike, 2017-03-11:
1. Stratux version: 1.2r1-0360e
2. Stratux config:

    SDR
      - [ ] single
      - [X ] dual

    GPS
      - [X ] yes
      - [ ] no
      type: VK-162

    AHRS
      - [ ] yes
      - [X] no

    power source: Anker E5

    usb cable: Anker

3. EFB app and version: iFly v9.8.7

    EFB platform: Andrioid KK

    EFB hardware: Asus Zenpad 

4. Description of your issue:

This isn't a code issue, per se. But something changed in the Stratux code that is making more RFI now. (I'm gonna guess it' the faster psuedo-AHRS.)

About a month ago, the squelches on both our COMS (King KX-165 and KX-155) started "breaking." (That it, started letting noise thru.) We noticed it on Tower (118.4 MHz), Ground Freq (121.8 MHz), Guard (121.5 MHz), Unicom (122.8). Whether it does it in the 130's, I don't know.

Tonight I had the brainstorm to reboot the Stratux in flight. As soon as I hit the reboot command, the noise in the COM stopped! When the Stratux came back on line, so did the noise.

This started in early February. So that would be coincident with the Feb 3 sh updates.

This is a Pi 3. No copper foil shielding like I had on my old Pi 2. Aside from the software, nothing has changed in my config. The Stratux is where it's always been. The antennas and SDR's have been where they've always been, etc.

I don't think that the developers can really code around EMI. More like the Project might start thinking about shielding the box if others report similar problems with their COMs.

cyoung, 2017-03-11:
I do think this may have to do with CPU load.

https://github.com/cyoung/stratux/blob/master/docs/RPi3/RA-029073-45-00B%20-%20part1.pdf
page. 22

A number of users have reported similar issues and it seems to be worse (according to reports) at ~125.0 MHz.

We were never able to figure out the source exactly, but it may be the time to start working on this.

Ergonomicmike, 2017-03-11:
Interesting. I quickly skimmed the report and I see the spike at COM freqs. (Correlates with what I happened on.) But I didn't see what they had the Pi doing for the test. Just powered up? Running Jessie? Calculating Primes? 

If it's the CPU radiating, then we ought to be able to shield some of this down. I've got copper foil at the hangar. But will be a least a week until I'm able to get out there again.

Actually, this sounds like something right up @Axtel4's alley.

Nokomis449, 2017-03-11:
Probably unrelated but I was getting similar noise on certain comm freqs, and finally narrowed it down to the 12v power cable leading to my iFly 740.  A trip to Radio Shack for a small choke fixed the problem.  They're cheap enough; might be worth a try.

Ergonomicmike, 2017-03-11:
I've got two simultaneous threads going on this issue (one here and one in Slack). In the Slack thread, I mentioned that, since I'm running my Stratux on an Anker, this particular problem is radiated. BUT, that doesn't mean that the Stratux isn't also conducting RF into ship's power for those who are hooked up. 

Did you tell iFly about the conducted EMI from the 740? They should have chokes internally on the power. 

Years ago I had to put a choke (also from Radio Shack) on an HID power supply in the Glasair.

Since then I thought Radio Shack declared bankruptcy. Are there still stores around?

Nokomis449, 2017-03-12:
It was AP that told me to try a choke.  Interference occurred both with ship's power and 12v portable battery.  Radio Shack filed for bankruptcy and closed a bunch of stores, but a few remain(ed) open.  It's been a few months since I've been to one, but the shelves were was pretty sad.

Ergonomicmike, 2017-03-12:
Interesting that a choke on ship's power fixed the problem even when you had the problem with battery power. Might point to what Chris found today, that moving his power wires around changed the squelch brake, So that might be telling us that the power cable is acting as an antenna. 

I've got a clip on choke from a computer cable I will put in my flight bag to test next time I'm at the hangar. Would be an easy fix that everyone could do. (Easier than soldering on a copper shield.)

cyoung, 2017-03-14:
If anyone else is experiencing this, give this config.txt a try (replace `/boot/config.txt` with the file, reboot). I'll be testing this further this week.

[config.txt](https://github.com/cyoung/stratux/files/841633/config.txt)

cyoung, 2017-03-17:
By the way, you can access the "boot" partition on a Windows/Mac/Linux and modify the config.txt file that way.

Ergonomicmike, 2017-03-17:
Yeah, I remember. I leave the Stratux in the plane nowadays, so have to SSH in there. I especially don't want to move anything until I've got a better handle on the RFI.

Ergonomicmike, 2017-03-18:
I did some testing in the hangar today with Comm 2 (King KX-155) on 125.0 MHz. I made a video.

You can hear the squelch breaks in the video as I do some "hands on" testing. 

There is a difference in the tone of the radiated RFI for the two different config.txt's. But the new config.txt does not solve the problem. The problem is radiated RFI, and it seems to be coming mostly from the controller chip. (Followed by the CPU and then something by the SD card.)

Here's the video: https://vid.me/gpZw

Jump to 3:20 for the final solution.

I didn't get a chance to fly it with the shielding today. But given how quite it is now, I'll be surprised if I have any more problems.

Note that this was with an Anker battery. (As opposed to ship's power.) I tried a clip-on choke first on the power lead. No difference that I noticed. (I'm gonna guess that, being battery based, the impedance looking into the Anker is very low.) Someone who uses ship's power should check for conducted EMI after they shield their Stratux.

cyoung, 2017-03-18:
In my testing with the new config.txt, the 125 MHz EMI was still present but there were less emissions at other COMM freqs. See the peaks around the left and right of 125 MHz here, these were eliminated on my setup. They appeared as the CPU frequency changed from 600 MHz to 1.2 GHz depending on CPU load. Raspbian is configured by default to use only these two frequency steps. 

As noted on Slack, your test raises the question as to how the aluminum heat sinks that some users put on the LAN chip change the EMI situation.

[image]

cyoung, 2017-03-18:
Here is a summary of the config.txt used over all releases. Most notable was the switch back to a floating CPU frequency with v0.8r2 (Jessie).

I suspect that this is related to #451.

| File                                              | Date     | Comment           | Config                           | Config Ident | CPU      | 
|---------------------------------------------------|----------|-------------------|----------------------------------|--------------|----------| 
| stratux-pre-8b4fea41f2-08252015.img.config.txt    | 8/25/15  | Wheezy. RPi2 only | 7dfb47e91e8a4b507f3f4921ec9c97a0 | A            | floating | 
| stratux-pre-a408a8cf13-08152015.img.config.txt    | 8/15/15  | Wheezy. RPi2 only | 7dfb47e91e8a4b507f3f4921ec9c97a0 | A            | floating | 
| stratux-pre-e17c0a5302-08312015.img.config.txt    | 8/31/15  | Wheezy. RPi2 only | 8aaf2de4fcf2c7f43bebc7a798663cfd | B            | floating | 
| stratux-v.0.1-08072015.img.config.txt             | 8/7/15   | Wheezy. RPi2 only | 7dfb47e91e8a4b507f3f4921ec9c97a0 | A            | floating | 
| stratux-v0.2r1-f47a224c1d-09042015.img.config.txt | 9/4/15   | Wheezy. RPi2 only | 8aaf2de4fcf2c7f43bebc7a798663cfd | B            | floating | 
| stratux-v0.3b1-7e61c7556a-09112015.img.config.txt | 9/11/15  | Wheezy. RPi2 only | 8aaf2de4fcf2c7f43bebc7a798663cfd | B            | floating | 
| stratux-v0.3b2-5a568e27ce-09162015.img.config.txt | 9/16/15  | Wheezy. RPi2 only | 8aaf2de4fcf2c7f43bebc7a798663cfd | B            | floating | 
| stratux-v0.3b3-c3e6d41ebe-09192015.img.config.txt | 9/19/15  | Wheezy. RPi2 only | 8aaf2de4fcf2c7f43bebc7a798663cfd | B            | floating | 
| stratux-v0.3b4-75f60562af-09242015.img.config.txt | 9/24/15  | Wheezy. RPi2 only | 773ea70f6708ceeff741ab812092621f | C            | floating | 
| stratux-v0.4r1-34f55beb04-10052015.img.config.txt | 10/5/15  | Wheezy. RPi2 only | 773ea70f6708ceeff741ab812092621f | C            | floating | 
| stratux-v0.4r2-a2a7b11494-10072015.img.config.txt | 10/7/15  | Wheezy. RPi2 only | 773ea70f6708ceeff741ab812092621f | C            | floating | 
| stratux-v0.4r3-6f5f69b-10202015.img.config.txt    | 10/20/15 | Wheezy. RPi2 only | 773ea70f6708ceeff741ab812092621f | C            | floating | 
| stratux-v0.4r4-293d0e74a8-10222015.img.config.txt | 10/22/15 | Wheezy. RPi2 only | 773ea70f6708ceeff741ab812092621f | C            | floating | 
| stratux-v0.5b1-ab889fb116-11212015.img.config.txt | 11/21/15 | Wheezy. RPi2 only | 773ea70f6708ceeff741ab812092621f | C            | floating | 
| stratux-v0.5b2-e6ad9aa830-12212015.img.config.txt | 12/21/15 | Wheezy. RPi2 only | 773ea70f6708ceeff741ab812092621f | C            | floating | 
| stratux-v0.5b3-680284ee1e.img.config.txt          | 12/27/15 | Wheezy. RPi2 only | 773ea70f6708ceeff741ab812092621f | C            | floating | 
| stratux-v0.5b4-d682bb5cf0.img.config.txt          | 1/1/16   | Wheezy. RPi2 only | 773ea70f6708ceeff741ab812092621f | C            | floating | 
| stratux-v0.5b5-cfd1541708.img.config.txt          | 1/7/16   | Wheezy. RPi2 only | 773ea70f6708ceeff741ab812092621f | C            | floating | 
| stratux-v0.5b6-1be927e672.img.config.txt          | 1/26/16  | Wheezy. RPi2 only | 1d76ea8d7359baf024a9a0fd951177b8 | D            | 900MHz   | 
| stratux-v0.6r1-bbbcd3bbe1.img.config.txt          | 1/29/16  | Wheezy. RPi2 only | 1d76ea8d7359baf024a9a0fd951177b8 | D            | 900MHz   | 
| stratux-v0.7b1-02bf2251d8.img.config.txt          | 2/15/16  | Wheezy. RPi2 only | 1d76ea8d7359baf024a9a0fd951177b8 | D            | 900MHz   | 
| stratux-v0.8r1-796d0a2632.img.config.txt          | 3/7/16   | Wheezy. RPi2 only | 1d76ea8d7359baf024a9a0fd951177b8 | D            | 900MHz   | 
| stratux-v0.8r2-db130aab76.img.config.txt          | 3/21/16  | Jessie. RPi2+3    | 7b1933cfa3adf9b9dfe68afebfd97476 | E            | floating | 
| stratux-v0.9b1-89107818b8.img.config.txt          | 5/27/16  | Jessie. RPi2+4    | d26f0c1655c39904674ed33f95975982 | F            | floating | 
| stratux-v0.9b2-64710fe60d.img.config.txt          | 6/21/16  | Jessie. RPi2+5    | d26f0c1655c39904674ed33f95975982 | F            | floating | 
| stratux-v1.0r1-945c8bf6bc.img.config.txt          | 7/13/16  | Jessie. RPi2+6    | d26f0c1655c39904674ed33f95975982 | F            | floating | 
| stratux-v1.1b1-0107b59010.img.config.txt          | 12/12/16 | Jessie. RPi2+7    | 794ca6d7ac129bbe22dd7d25075fd4de | G            | floating | 
| stratux-v1.2r1-9ee46170ff.img.config.txt          | 1/29/17  | Jessie. RPi2+8    | 794ca6d7ac129bbe22dd7d25075fd4de | G            | floating |

merricke, 2017-09-25:
My solutions for EMI and power success:

First of all, thanks Chris and other contributors for the Stratux project.  I’m one of the happy beneficiaries.
I’m a newby, not sure if any of this is new to the group.

Conclusion:
EMI was brought under control by grounding the RPi heat sinks and also adding a ‘rat’s nest’ grounded random wire bundle over the Pi board.
Power supply is with a dual port usb car charger, Aukey CC-S7, and powered hub, Anker Ultra Slim USB hub w/USB charger.  The Pi goes to one port of the charger while the hub gets power from the other port.

Story:
My objective is to not use a battery supply.
Adequate power for the Pi with the car chargers was a problem with peripherals attached (VK-62 and one SDR).  The Pi was running at the low end (red light blinking) using various power sources.  I tried:
1.	Cheap dual port car usb charger, DC/DC converter but inefficient and put noise on Pi carrier
2.	Cheap adjustable DC/DC converter board.  Provided good power but very noisy.
3.	Made a linear supply.  Clean but hot.
4.	Anker car charger.  Clean, Aukey slightly higher output under load.
5.	Aukey car charger.  Clean, power adequate but marginal, about 4.7 v on Pi bus.
6.	Added Anker powered usb hub for peripherals.  4.9 v on Pi bus.

Occasionally I suffered SD card corruption during shutdown with the marginal PS’s, so I had to re-image the card a few times.  Haven’t had that problem lately.

EMI problems were from two sources.  One is the 125.0 carrier from the Pi.  Other is information being impressed on the Pi’s carrier by noisy power supplies.  With enough modulation and a strong carrier, all communications were being impacted by Stratux.  Using a clean PS eliminated the transmitted noise information, and grounding the Pi’s heat sinks brought the Pi’s carrier power down enough to avoid spurious breaks of com squelch.  

My testing on the bench was crude, just used a handheld radio.  The results provided enough contrast that I didn’t bother putting any components on a scope.

Discovered by accident that a 3’ piece of loose signal wire attached to the ethernet can and the cpu heat sink also attenuated the Pi carrier.  Tried wrapping board in metal screen and using longer wire but no gain was noticed over just jumbling the 3’ wire on top of the Pi.

I also observed noise from the SDR on 120.0, however, it wasn’t strong enough to trip my squelch.  I don’t know if that would interfere with a 120.0 com channel.

My setup is old, analog King KY-197 com’s, RPi3, VK-62 GPS, and thumb drive size SDR (from Vilros, summer of 2016).  My screen is a Windows laptop running the Ifly app.

Ergonomicmike, 2017-09-25:
3 foot or 3 inch? Got any photos?

merricke, 2017-09-25:
3 feet.  I’ll try to remember to get some pix Wed.  Can you put pix on Github or do you put them on another site?

Ergonomicmike, 2017-09-25:
I've left images before. I think you just click on "Attach files" below.

merricke, 2017-09-28:
Pix of the grounded heat sinks & jumbled wire (emi shield).
I used solid hookup wire which is 'haywire twisted' to one heat sink and deformed to make sufficient mechanical contact with the other heat sink.
The gray wire is 28ga stranded wire.

[image]
[image]

Ergonomicmike, 2017-09-29:
Interesting. I expect that the grounded shields are coupling the EMI from the chips to ground. (Earlier, in a video somewhere, I had put my finger over the chips (without heat sinks) and found that my finger radiated away energy, making more noise.) If so, then it might be that short braid would also do the trick too.

merricke, 2017-09-29:
Depending on where you're doing your tests, your body may be adding 60 Hz noise to the system.  Try using clip leads to ground the heat shields to the USB or Ethernet cans to see if you get similar results to mine.
I think it's a 2 step process:
-reduce the Pi's 125.0 carrier strength with EMI shielding (grounding the heat sinks & the 'wire cage')
-reduce modulated information (below 8 KHz for my KY197) on the carrier that opens the squelch circuit by using a clean power supply.

I'm assuming battery supplies also use DC/DC converters so they may not necessarily be noise free.  I didn't test with any battery supplies.   And adding your own filters (capacitors or chokes) my alter the output characteristics of the converter instead of smoothing the current.

GitUser275, 2017-11-20:
As a new Stratux owner I've been browsing some of the old discussions.  I found this one interesting so I ran an EMI current probe over my unit.  I found a signal at 125.01 MHz with the probe centered over the micro SD card (marker 1).  There were also signals at 115.25 MHz and 120.08 MHz when the probe was over the NooElec RTL receivers (markers 2 and 3).    I don't know yet if these signals will cause any problems with my aircraft radios.  I didn't find any other signals as strong as these.

In the images the yellow trace is with Stratus powered down.  The magenta trace is with the current probe over the micro SD card.  The cyan trace is with the current probe  above the  NooElec RTLs.

[image]
[image]

merricke, 2017-11-20:
That's good info, and it concurs with my crude tests just using a handheld radio.
I'm wondering, at 120 I was getting a high pitch squeal, maybe from the 2 carriers?

Ergonomicmike, 2017-11-28:
Very cool @GitUser275 ! The 115.25 MHz explains the noise I'm getting on 116.0 MHz. Next time I do a video (below) I'll go down to 115.25.

Can you also scan into the 300 to 400 MHz? We're beginning to wonder about GS frequencies. And since there's no way to listen to what's being received on the GS, it would be good to know what's there at fundamental frequencies. Also, please scan from 130 up to 300 - might be something in the 160's who's harmonic could interfere with GS.

In the meantime, I found that some of this noise is being conducted into the power cord and then radiated out. See my video at https://vid.me/ADfa2

I'm in the process of putting braid over the power cord and also the USB cable for the 1090 SDR.

GitUser275, 2017-11-28:
For the spectrum traces shown above both SDR were active (RX 978 and RX 1090).   I tried unplugging one to see which one was causing which frequency but I don't have a capture.   I'll run some more tests and report back.

Snap on ferrite beads can be quite effective at suppressing spurious VHF signals on power and signal cables. Put them as close to the source as possible.

Are you at DVT?  I recognized the frequencies in the first post of this thread.

Ergonomicmike, 2017-11-29:
@GitUser275 Yes, DVT. Per the video, tried a snap on bead. I couldn't tell a difference.(Maybe I don't have the right ones for the freqs we're talking about. @cyoung reports success with his.

Ergonomicmike, 2017-11-29:
Also, doing testing in a Glasair. Not having a metal shell for any shielding, ours is the Worst Case airplane for EMI problems.

GitUser275, 2017-11-29:
DVT, Mike, and a Glassair?  Maybe I know you.  J3 at Sunburst and flight review 6/19/2007??

Andy

Ergonomicmike, 2017-11-29:
That's me. (Pretty scary, huh?) I keep track of time nowadays by the Glasair. So, we finished it 1995. I thought that Sunburst had been forced to shut down some time back then. (As opposed to only 10 years ago in 2007.)

GitUser275, 2017-11-29:
it's a long time ago but you checked me out to solo in Jim's J3 at Sunburst and much later in 2007, long after Sunburst had closed, I gave you a flight review in your bird.  If you think heel brakes were bad - I think I went grey instructing in the right seat of an Aeronca Chief.  No brakes on that side.

It's a small world.  Good to hook up again in a new area of interest.

Ergonomicmike, 2017-11-29:
Ahhh. So Andy D? (of Glider fame?)

GitUser275, 2017-11-29:
Yes, that's me.  Moving back on topic I made the requested sweep using the EMI current probe.

For 300-400 MHz I only found very weak signals near the micro SD card slot:

[image]
[image]

For 130-300 MHz the significant signals were in the vicinity of the two RTL SDR:
[image]
[image]

In both cases the yellow trace is a baseline with Stratux unpowered.  These traces use power average sampling with 100 sample average.  Note that the amplitude scale is not the same as for the earlier traces.

Ergonomicmike, 2017-11-29:
Are you doing this at work? Or do you have your own analyzer? If the latter, and if you're still around DVT, maybe we can sniff out my unit in the plane?

GitUser275, 2017-11-29:
My retirement hobby was to renew my interest in amateur radio and the spectrum analyzer was bought for that.   Sure, we could meet some time at your hangar.  Is your aol email still valid?

Ergonomicmike, 2017-11-30:
Yep. 

I sold my old Heathkit HF equipment at a yard sale for $100 a few years ago. (Obsolete I guess, since they didn't have the new HF bands that came in later.)

merricke, 2017-12-01:
GU275, can you run the scans with some test leads grounding the heat sinks to the board ground?  I'm wondering if you'll get more definitive results than I did at 125.0.

GitUser275, 2017-12-01:
As I said earlier - " I found a signal at 125.01 MHz with the probe centered over the micro SD card (marker 1). "   If I move my 1.5 inch diameter probe away from the card holder the 125.01 MHz signal is very quickly attenuated.   I detected no significant signals with the  probe centered over either heat sink.   

Later I'll see if the 125.01 peak is influenced by grounding the heat  sinks but it may be tricky to get good contact.  They appear to be anodized aluminium which is not conductive unless the anodizing is removed.

The probe I'm using is a home brew 1.5 inch diameter shielded magnetic field probe.

GitUser275, 2017-12-03:
I looked again with the probe close to each heat sink and I find no strong signals there.  There is a very strong 250 MHz signal with the probe centered over the RAM chip which is other side of the board.   Here are a couple more screen captures.

Centered on RAM:
[image]
(marker table not captured)

Centered on micro SD card holder:
[image]
[image]

cyoung, 2017-12-03:
@GitUser275 - regarding the 115.25 MHz and 120.08 MHz signals, do these vary with input voltage (to the Pi)?

merricke, 2017-12-03:
@GitUser275 , I'm wondering
- the area of noise around 103 MHz on both sides of the board
- the periodic spikes about every 42 MHz
- would the signal at 125 over the SD card drop off if the heat sinks are grounded (mine aren't alodined and are easy to connect to with test leads).  My way of checking signal strength was to check squelch requirements at differing distances with my handheld radio (done at home but improvement there matched improvement in the plane).

Ergonomicmike, 2017-12-03:
While flying yesterday, ran thru the NAV channels. 

This is not an exhaustive list, but to my surprise, found noise on:

108.00 (Some kind of Intermod. Could hear talking. Might not be Stratux.)
108.55
109.25
109.40
110.00
110.15
110.70
111.15
111.95
112.00
112.05
112.85
112.90
112.95
113.75
114.60
114.65
114.70
115.20 (Solid, like a carrier. None of that repetitive noise associated with the SDR.)
115.60
116.40
116.45
117.30
117.35

GitUser275, 2017-12-03:
I made a much smaller probe.  This one has a loop diameter about 0.5 inch.  Using it I was able to explore the processor side of the board.   I found the strongest signals in the are equidistant between each heat sink and and the header, an area with no large devices.   I'll do more more later in the week. 

[image]

Nokomis449, 2017-12-03:
It'd be nice if someone could line their case with copper tape and see if it reduces the noise.
https://www.amazon.com/dp/B07663WM7D/ref=sspa_dk_detail_3?psc=1&pd_rd_i=B07663WM7D&pd_rd_wg=EpeiX&pd_rd_r=BX062CXSD8E3WZ0NK9M6&pd_rd_w=az1aW

GitUser275, 2017-12-03:
Re - voltage influence 

What range of voltage would be of interest.   I'm running off a bench supply but via a small adjustable regulator board that has been carefully tweaked to 5.25 V.  I did that so I can't overvolt the Pi.

cyoung, 2017-12-03:
4.65V - 5.25V or thereabouts.

merricke, 2017-12-04:
My frequency notes, I'm only using 978 SDR, not 1090
108.35- static, localizer
115.20- carrier, VOR, from SDR, matches with GU275
115.52- static
120.00- carrier w/high pitch noise, ATC, from SDR, matches w/GU275
125.00- carrier, ATC, RPi, matches w/GU275
129.60- carrier w/static, from SDR
129.87 to 129.95- static
131.12 to 131.15- static
131.60 to 131.70- static

GitUser275, 2017-12-04:
I set out to see if the levels of the 115 and 120 MHz signal were voltage dependent but was distracted by something else.

The 115 and 120 Mhz signal pair is localized to the RTL SDR dongles.  This pair is present when either RTL is plugged in and when both are plugged in.   The pair first appears after power application just before the fan test runs.   These levels are shown in the amber trace.  1 minute and 42 seconds after power application the  level of the 115 MHz signal rises as shown in the magenta trace.    

What causes this change in signal level?

[image]
[image]

Axtel4, 2017-12-04:
There is a 90 sec delay for the R-Pi start up to the activation of the SDRs to give the GPS a chance to lock.  Does the SRD levels return to the lower value when they are turned off in the WebUI?

Axtel4, 2017-12-04:
@merricke Is that data with or without your "spaghetti" fix?

GitUser275, 2017-12-04:
"Does the SRD levels return to the lower value when they are turned off in the WebUI?"

No, the levels stay constant at the high level when both 1090 and 978 are selected to off.  However, if Stratux is re-booted with both selected off, the 115.2 MHz signal stays at the low level. The level rises at each  SDR when that SDR is later selected to on.

Axtel4, 2017-12-04:
@GitUser275 Interesting that it stays the same level when the SDR is turned off in the IU.  I suspect the 115.2 MHz signal is the forth harmonic of the 28.8 MHz SDR master clock.  The 125 MHz signal is the clock frequency of the USB/Ethernet chip on the Raspberry Pi.

rhole, 2017-12-04:
It is possible that turning the SDR off disables only part of the device and the master oscillator may remain on. You could physically remove the SDR to find out.
Also know that a good DC ground is not necessarily a good RF ground. I would provide a ground wire (grounding braid is best) 14g or better if there is no metal airframe to use.
I’ve seen many noise issues with, for example, handi-talkies greatly improved with a good ground.

Rick  Hole

merricke, 2017-12-05:
@Axtel4 Without the wire cage.  I concentrated on getting the Pi's 125 carrier attenuated, never made a comprehensive test later.  I saw a significant improvement when grounding the heat sinks and another drop with the wire rat's nest.  My plane's radios are back to normal, not breaking squelch and no extraneous noise on the freq's I use.
The early power supplies I had were a main source of noise, I believe.  With the better power supplies I haven't experienced any noise problems.

GitUser275, 2017-12-05:
"No, the levels stay constant at the high level when both 1090 and 978 are selected to off."

I repeated the test this morning.  I see that the signal at the 978 SDR reduces when it is selected off with the GUI.  The signal at the 1090 SDR does not reduce when it is selected off with the GUI.    I did repeat the test several times this morning, with and without a GPS connected, and the results are consistent.  978 signal is controlled by UI on/off but 1090 signal is not controlled by UI on/off.

As observed yesterday - if both are selected off, then the Stratux is booted, the low level signal is seen at both SDR.

I suspect my observations of yesterday were invalid because the GPS USB connector prevented close positioning of the probe over 978 SDR.  In today's test I moved the GPS to a different USB port to allow close contact of the probe with either SDR.

Axtel4, 2017-12-06:
@GitUser275  Have you happened to connect the SA directly to the SDR antenna port and swept the frequency band?  Also tried connecting the SA to the antenna port through a directional coupler and a Sig gen to the coupled port with injecting an in band signal and sweep the frequency band?

GitUser275, 2017-12-06:
"Have you happened to connect the SA directly to the SDR antenna port and swept the frequency band? Also tried connecting the SA to the antenna port through a directional coupler and a Sig gen to the coupled port with injecting an in band signal and sweep the frequency band?"

Simple answer to both - No.

I'm willing to investigate what is transmitted by each SDR but you'd need to specify what frequency range you are interested it.

I don't understand your proposed use of the directional coupler.  Typically an SA with TG and directional coupler are used to characterize S11 of an antenna system and I have made such measurements on my station antennas.  Why would you want to connect the coupler to the SDR antenna port?  What are you expecting to measure?

cyoung, 2017-12-06:
Here's a sweep up to 3.2 GHz with the SA connected directly to the antenna ports, as suggested by @Axtel4. v1.4r3, dual band unit with internal GPS, in the white ABS case ("pre-built" configuration). This is important since the pre-built config has the antenna feed looped over the top of the Pi, so there may be some noise coupled onto the antenna feed from that (maybe the Wi-Fi stuff).

Magenta is baseline (Stratux off, SA input not connected to Stratux). Blue is peak hold while connected to the port and Stratux on (with uptime 2 minutes+). There's a peak at 345.6 MHz from something outside of Stratux, not sure what.

1090 port:
Peaks at
85.33 MHz @ -85.83 dBm
115.20 MHz @ -84.95 dBm
230.40 MHz @ -81.82 dBm
345.60 MHz @ -78.71 dBm (external noise)
1540.27 MHz @ -79.82 dBm
1902.93 MHz @ -80.04 dBm
2184.53 MHz @ -69.04 dBm
2410.67 MHz @ -65.25 dBm

[image]

978 port:
85.33 MHz @ -85.83 dBm
115.20 MHz @ -84.77 dBm
230.40 MHz @ -81.82 dBm
345.60 MHz @ -78.71 dBm (external noise)
1540.27 MHz @ -79.57 dBm
1902.34 MHz @ -80.04 dBm
1958.40 MHz @ -75.69 dBm
2184.53 MHz @ -69.04 dBm
2410.67 MHz @ -65.25 dBm

[image]

Axtel4, 2017-12-06:
@GitUser275 
Looking at the output of the SDR antenna port will show if there is any LO leakage or oscillations being emitted from the SDR.  Stray signals radiating for the antenna port would require other mitigation efforts that shielding the SDR.  

Using the directional coupler and the external generator allows the measurement of the generation of intermod products at the front end of the SDR in the presence of strong nearby signals.  For example, if the SDR has ESD clamping diodes at the antenna port, a strong signal entering the antenna port could cause the creation of intermod distortion across the PN junction of the clamping diode.  I've also seen radiated intermod products created at the output of receiver preamps in the presence of strong signals.  

A better method would be to inject the signal though a circulator, but most home labs don't have one available so a directional coupler is substituted.

Axtel4, 2017-12-06:
@cyoung 

Thanks.  That is an interesting look.  As stated above, one of the potential interference sources is unintentional radiated emissions from the antenna port.  The 115.2 MHz signal appears to be a reoccurring theme.  I wonder if the received noise at that frequency is affected if a dummy load is placed on the SDR antenna port?

GitUser275, 2017-12-07:
@Axtel4 

I understood why you wanted the antenna port scan but was not aware of that technique for evaluation of front end intermod.   I'll try to find some references.   Maybe you can point me to some? 

I don't think  I want to send my time evaluating RTL based receivers though.  I don't expect much for $15.

merricke, 2017-12-08:
@GitUser275 Dirty world we live in; my wifi router puts out a carrier at 120.
Interesting that your grounding of the heat sinks had different results from mine: puzzling.
Which components seem to benefit from the 'finger' grounding?

Ergonomicmike, 2017-12-10:
(Cross posted this in Slack.)

I did the (mostly) shielded Ammo Box text yesterday. Yes, there is noise being radiated from the antennas. That was the best case test for a Stratux. Everything in the box (Pi, battery, SDR's) with only 978, 1090 and GPS antenna outside.

BTW, it seems that the GPS cable is conducting and then radiating noise. So I might have to braid that.

Otherwise, with the braid on my power cable and most of the 1090 SDR, most of the COM noise is gone. 

Now that the noise floor has dropped a bit, I'm finding that the 978 SDR is causing trouble. So am going to copper foil that today before flying tomorrow.

[10 more replies]