cyoung/stratux #574: WiFi disconnecting

N9983y, 2017-03-18:
Hi, if I turn on with dual sdr plugged in, I get a blinking red light.   If I power on without sdr plugged in it's fine.  But as soon as I plug in sdr, power flash and loose wifi and red light start blinking.  I've tried with multiple power source and cables but same problem.  Also formatted card and reloaded stratus.  

1. Stratux version:
2. Stratux config:

    SDR
      - [ ] single
      - [ ] dual

    GPS
      - [ ] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [ ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)

    EFB hardware: (e.g., iPad Mini 2)

4. Description of your issue:

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

N9983y, 2017-03-18:
Also the lights on the sdr are flashing.  If I boot without sdr, let it boot, wait and plug sdr 1 by 1, sometimes it will work.

cyoung, 2017-03-18:
Thanks for the information. Sounds like you have a power supply issue. Try a shorter cable and/or different powerbank.

cyoung, 2017-04-19:
Please update if you continue to have the issue. You may want to upgrade to v1.2r2.

closed 2017-04-19 by cyoung