cyoung/stratux #576: becomes unresponsive... stratux log shows loopint on GPS attitude: Current GPS time is older...

bglaplante, 2017-03-26:
1. Stratux version: 1.2r1
2. Stratux config:

    SDR
      - [ x ] single
      - [ ] dual

    GPS
      - [ x ] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [ x ] no

    power source: AC adapter

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2)

    EFB platform: (e.g., iOS 9.2)  n/a

    EFB hardware: (e.g., iPad Mini 2)  n/a

4. Description of your issue:
After several days of running on the bench, Statux will go unresponsive.  Stratux.log shows repeated messages 
GPS attitude: Current GPS time (0.20) is older than last GPS time (86400.80).  Checking for 0000Z rollover.

The messages then repeat with Current GPS time incrementing...

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

cyoung, 2017-04-19:
Looks like it may be a problem with using the GPS timestamp when calculating GPS attitude. I think this should change to using `stratuxClock` instead of using `myGPSPerfStats[...].nmeaTime`. It seems valid to assume that if the GPS produces a valid fix position over serial that it was obtained within the last serial update cycle? Or was this an attempt to circumvent coasting? @AvSquirrel

cyoung, 2018-01-17:
I don't remember this getting fixed, but couldn't reproduce.

closed 2018-01-17 by cyoung