cyoung/stratux #577: SkyDemon Support

crablab, 2017-03-28:
1. Stratux version:
2. Stratux config:

    SDR
      - [* ] single
      - [ ] dual

    GPS
      - [* ] yes
      - [ ] no
      type:

    AHRS
      - [ ] yes
      - [ *] no

    power source: LiPo via regulator and soldered directly to voltage rails

    usb cable: N/A

3. EFB app and version: SkyDemon 3.7.1

    EFB platform: Android 7.1.1

    EFB hardware: Nexus 9

4. Description of your issue:

There doesn't seem to be any information on whether SkyDemon should/will be able to connect to Stratux. SkyDemon supports data in via FLARM, Dynon SkyView, X-Plane and GDL90 - does Stratux support any of these protocols? There is a brief mention that by changing the IP of Stratux to 192.168.1.1 you can get SkyDemon to work with Flarm but this was never fully described: http://forums.skydemon.aero/Topic19950.aspx

crablab, 2017-03-28:
Note: for use in Europe, RunwayHD and SkyDemon are the *only* applications really.

cyoung, 2017-04-19:
@crablab - there is some commentary here on using SkyDemon with Stratux:
https://github.com/cyoung/stratux/issues/489

Stratux outputs data in GDL90 format to port 4000 to all Wi-Fi clients.

closed 2017-04-19 by cyoung