cyoung/stratux #579: Pi Zero W

crablab, 2017-03-29:
1. Stratux version: Latest
2. Stratux config:

    SDR
      - [-] single
      - [ ] dual

    GPS
      - [-] yes
      - [ ] no
      type: VK162

    AHRS
      - [ ] yes
      - [-] no

    power source: 2A stabilised PSU

    usb cable: Anker

3. EFB app and version: (e.g., WingX Pro7 8.6.2)
SkyDemon 3.7.1
    EFB platform: (e.g., iOS 9.2)
Android 7.1.1
    EFB hardware: (e.g., iPad Mini 2)
Nexus 9
4. Description of your issue:
On install, everything is great - WiFi network comes up and everything connects and works. The first time the device is powered down (via the web interface), on the next boot the wireless network fails to come up. First time this happened I thought I'd mucked up so re-imaged the SD card, reinstalled Stratux and again, everything was fine until I rebooted. 

Debugging, it appears ACS fails stopping the adaptor from ever starting up: 
[image]

crablab, 2017-03-30:
So, re configuring the AP manually works fine - it looks like this is a specific problem with Stratuxs' implementation. Perhaps with the script it runs after the AP is up?

cyoung, 2017-03-30:
How'd you reconfigure it?

cyoung, 2017-04-19:
https://github.com/jpoirier/stratux-setup for Pi Zero W, or comment here to let us know how you got it working.

closed 2017-04-19 by cyoung

crablab, 2017-04-20:
I had tried that link previously - it doesn't work. 

I used an alternative version, which from memory was a FLARM fork although I'll have to find that version. I compiled manually and didn't compile the UAT library (as it's useless in the UK).

crablab, 2017-04-20:
I used this fork: https://github.com/AvSquirrel/stratux/tree/flarm-network
And compiled following these instructions: https://github.com/AvSquirrel/stratux/tree/flarm-network

I set up the wireless network manually however, although I can get the config pages I can't get SkyDemon to connect to it.

flyingyanz, 2017-08-01:
Any updates on this issue?

I was trying to compile the latest stratux from here, and having problems with wiringpi.

It complains cannot find <wiringPi.h>, and then about ld -lwiringpi fail.

All of those problems are after I did, still, it compains the above problems
pip install wiringpi
apt-get install wiringpi

crablab, 2017-08-02:
I just gave up!