cyoung/stratux #580: GDL-90 traffic alert status

semaphore-digital, 2017-03-30:
I've been flying with stratux connected to Foreflight, and also pushing data to my Advanced Flight 4500s EFIS via GDL-90 serial.  Everything's working well and I've done a [writeup](http://www.vansairforce.com/community/showthread.php?t=148044) to share with other pilots with similar config; this is one of a few dual-band In solutions that work with this hardware, and by far the most affordable.

In the build and testing, I did have one issue that required a change & recompile of the stratux code.  Because the GDL-90 "Traffic Alert Status" byte is hard-coded as 1 ("indicates that a traffic alert is active for this target" per the GDL-90 spec), the EFIS displayed all received targets in alert status.

Changing 
`msg[1] = 0x10 | ti.Addr_type` ([1.1b1 at 248](https://github.com/cyoung/stratux/blob/0107b59010f5b816833a60c3e6b3e5ad07f6e1ec/main/traffic.go#L248)) 
to set the alert status to 0, as:
`msg[1] = 0x00 | ti.Addr_type` 
allowed the EFIS to handle the alert determination on its own; on my particular display this shows distant traffic as a hollow diamond, nearby traffic as a filled diamond, and alert traffic as a yellow circle.

I tested Foreflight 8 and a WingX trial with the status set to 0, and it didn't seem to make any difference in display there.  I can't speak to how other EFISs or EFBs handle the difference, though.

cyoung, 2017-03-30:
Thanks for doing initial testing on this. We'll have to try with a few other EFBs to see if they even pay attention to the bit.

ChiefPilot, 2017-04-03:
I've seen this same issue with certified equipment (Avidyne IFDs) and from reading the GDL90 ICD it seems this flag should only be set when the MFD is supposed to show a traffic alert.   Pull request to replace the hardcoded bit with logic that examines distance to target is inbound and conditionally sets the alert flag is inbound.

cyoung, 2017-04-06:
Needs EFB testing.

http://updates.stratux.me/builds/update-stratux-v1.2r2-9e12f69318.sh

Ergonomicmike, 2017-04-10:
@cyoung What do I look for on my EFB?

ChiefPilot, 2017-04-11:
@Ergonomicmike I'm sure @cyoung will chime in, but as the author of the commit that was pulled in, I'd be very interested to see if there was _any_ change at all with the way EFB displays traffic.   The hypothesis is that there will be no changes in how an EFB displays traffic, as the EFBs seem to ignore the flag entirely  (if they had been following the suggestions outlined in the GDL90 ICD, each target would have been displayed using a flashing yellow traffic symbol).

Ergonomicmike, 2017-04-12:
Okay, thanks. I'll let you know if I notice any changes.

cyoung, 2017-05-17:
No noticeable difference on any of the major EFBs so far. Closing until we run into an issue with it.

closed 2017-05-17 by cyoung