cyoung/stratux #583: Help, Possibly software issue?!

raylow78, 2017-04-06:
1. Stratux version: **v1.2r1**
2. Stratux config:

    SDR
      - [ ] single
      - [x ] dual

    GPS
      - [ ] yes
      - [ x] no
      type:

    AHRS
      - [ ] yes
      - [ ] no

    power source:**EasyAcc 6000mAh Ultra-Slim External Battery** 

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2) **ForeFlight 9.0**

    EFB platform: (e.g., iOS 9.2) **iOS 10.3.1**

    EFB hardware: (e.g., iPad Mini 2) **iPad Mini 4**

4. Description of your issue:

I just put together my dual band stratux using the items listed on the website. The only thing I did not purchase was the optional GPS because my iPad has an internal one built it. Once I powered it up and loaded ForeFlight nothing happens. The device powers on with a red light and steady green light, it does not blink. ForeFlight picks up that it is connected to an ADS-B however it will not display any traffic, weather or ads-b towers. I can hear airplanes flying overhead so I know they are there lol.

I don't think the software is loading properly onto the micro sd. I am running pi filler on my mac (OS X El Captain 10.11.6). I have unzipped the file before running it through pi filler. It says that all of the data is loaded on and to go ahead and plug it into the motherboard. However, if I place the micro sd back into my computer it shows the original data on the card rather than the statux software.

What should I do to troubleshoot?

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

2016/02/26 01:18:36 Stratux v1.2r2 (bcd62e7c00e132dada7ed7cbf0a5d28133916ff1) starting.
2016/02/26 01:18:36 Entered uatReader() ...
2016/02/26 01:18:36 can't read settings /etc/stratux.conf: open /etc/stratux.conf: no such file or directory
2016/02/26 01:18:37 ===== UAT Device Name  : Generic RTL2832U OEM =====
2016/02/26 01:18:37 ===== UAT Device Serial: stratux:978=====
2016/02/26 01:18:37     GetTunerType: RTLSDR_TUNER_R820T
2016/02/26 01:18:37     SetTunerGainMode Successful
2016/02/26 01:18:37     SetTunerGain Successful
2016/02/26 01:18:37     GetTunerGain: 480
2016/02/26 01:18:37     SetSampleRate - rate: 2083334
2016/02/26 01:18:37     GetSampleRate: 2083334
2016/02/26 01:18:37     GetXtalFreq - Rtl: 28800000, Tuner: 28800000
2016/02/26 01:18:37     SetXtalFreq - Center freq: 28800000, Tuner freq: 28800000
2016/02/26 01:18:37     SetCenterFreq 978MHz Successful
2016/02/26 01:18:37     GetCenterFreq: 978000000
2016/02/26 01:18:37     Setting Bandwidth: 1000000
2016/02/26 01:18:38     SetTunerBw 1000000 Successful
2016/02/26 01:18:38     ResetBuffer Successful
2016/02/26 01:18:38     GetFreqCorrection: 0
2016/02/26 01:18:38     SetFreqCorrection 0 Successful
2016/02/26 01:18:38 Entered UAT read() ...
2016/02/26 01:18:38 ===== ES Device Serial: stratux:1090 PPM 0 =====
2016/02/26 01:18:38 Entered ES read() ...
2016/02/26 01:18:38 Executed /usr/bin/dump1090 successfully...
2016/02/26 01:18:41 readBMP180(): input/output error
2016/02/26 01:19:06 client connected: 192.168.10.10:4000 (Raymonds-iPad).
2016/02/26 01:19:06 stats [started: 30 seconds ago]
2016/02/26 01:19:06  - Disk bytes used = 1.4 GB (76.4 %), Disk bytes free = 440 MB (23.6 %)
2016/02/26 01:19:06  - CPUTemp=36.48 deg C, MemStats.Alloc=2.3 MB, MemStats.Sys=6.6 MB, totalNetworkMessagesSent=0
2016/02/26 01:19:06  - UAT/min 0/0 [maxSS=0.00%], ES/min 179/179, Total traffic targets tracked=1
2016/02/26 01:19:06  - Network data messages sent: 0 total, 0 nonqueueable.  Network data bytes sent: 0 total, 0 nonqueueable.
2016/02/26 01:19:06  - Last GPS fix: 30 seconds ago, GPS solution type: 0 using 0 satellites (0/0 seen/tracked), NACp: 0, est accuracy 999999.00 m
2016/02/26 01:19:06  - GPS vertical velocity: 0.00 ft/sec; GPS vertical accuracy: 999999 m
2016/02/26 01:19:36 client connected: 192.168.10.11:4000 (Raymonds-MBP).
[... 70 lines cut ...]
2016/02/26 01:18:12     SetTunerGain Successful
2016/02/26 01:18:12     GetTunerGain: 480
2016/02/26 01:18:12     SetSampleRate - rate: 2083334
2016/02/26 01:18:12     GetSampleRate: 2083334
2016/02/26 01:18:12     GetXtalFreq - Rtl: 28800000, Tuner: 28800000
2016/02/26 01:18:12     SetXtalFreq - Center freq: 28800000, Tuner freq: 28800000
2016/02/26 01:18:12     SetCenterFreq 978MHz Successful
2016/02/26 01:18:12     GetCenterFreq: 978000000
2016/02/26 01:18:12     Setting Bandwidth: 1000000
2016/02/26 01:18:12     SetTunerBw 1000000 Successful
2016/02/26 01:18:12     ResetBuffer Successful
2016/02/26 01:18:12     GetFreqCorrection: 0
2016/02/26 01:18:12     SetFreqCorrection 0 Successful
2016/02/26 01:18:12 Entered UAT read() ...
2016/02/26 01:18:12 ===== ES Device Serial: stratux:1090 PPM 0 =====
2016/02/26 01:18:12 Entered ES read() ...
2016/02/26 01:18:13 Executed /usr/bin/dump1090 successfully...
2016/02/26 01:18:16 readBMP180(): input/output error
2016/02/26 01:18:41 stats [started: 30 seconds ago]
2016/02/26 01:18:41  - Disk bytes used = 1.4 GB (76.4 %), Disk bytes free = 439 MB (23.6 %)
2016/02/26 01:18:41  - CPUTemp=36.48 deg C, MemStats.Alloc=1.6 MB, MemStats.Sys=6.6 MB, totalNetworkMessagesSent=44
2016/02/26 01:18:41  - UAT/min 0/0 [maxSS=0.00%], ES/min 0/0, Total traffic targets tracked=0
2016/02/26 01:18:41  - Network data messages sent: 34 total, 34 nonqueueable.  Network data bytes sent: 582 total, 582 nonqueueable.
2016/02/26 01:18:41  - Last GPS fix: 30 seconds ago, GPS solution type: 0 using 0 satellites (0/0 seen/tracked), NACp: 0, est accuracy 999999.00 m
2016/02/26 01:18:41  - GPS vertical velocity: 0.00 ft/sec; GPS vertical accuracy: 999999 m

Helno, 2017-04-07:
Based on that log file it is definitely working. It appears to have seen traffic and is seeing your ipad and MacBook pro.

Take a look in the Foreflight settings and be sure that hide distant traffic is turned off.

raylow78, 2017-04-10:
I was able to figure out the issue...long story short it was a pi filler issue.

closed 2017-04-10 by raylow78