cyoung/stratux #584: Need Flarm module of AvSquirrel over release 1.2r2

ecsaulnier, 2017-04-18:
1. Stratux version: 1.2r2
2. Stratux config: bc54533

    SDR
      - [ ] single
      - [X ] dual

    GPS
      - [X ] yes
      - [ ] no
      type: VK-162

    AHRS
      - [X ] yes
      - [ ] no

    power source:

    usb cable:

3. EFB app and version: (e.g., WingX Pro7 8.6.2) Air Navigation Pro Last Version

    EFB platform: (e.g., iOS 9.2) IOS 10.3

    EFB hardware: (e.g., iPad Mini 2) iPhone 6 +

4. Description of your issue: 
Hello, 

For my EFB App "Air Navigation Pro" i need Flarm module of AvSquirrel. 

How install this over Stratux version 1.2r2 ?

Thank for your response. 
French User.

If possible, enable "Replay Logs", reproduce the problem, and provide a copy of the logs in http://192.168.10.1/logs/stratux/ and http://192.168.10.1/logs/stratux.log.

peterclubroad, 2017-04-18:
Yes please, it would be great to have that. I reckon that this would broaden the range of EFB listening to the data stream quite a bit. Thanks for any hint if AvSqirrels code can simply be used.
Regards
Peter, Germany

peterclubroad, 2017-04-27:
Found myself at the beginning of the learning curve. Tried to update 1.2R1 with AvSquirels Flarm-Update and tried to update 1.2R2 with the Flarm-Update. Could connect to Pi3 with Terminus afterwards but PI3 was no accessible through 192.168.10.1 anymore (started to load the page but freezed about 1/3 through).
A fresh version of 1.2R2 (bcd62...) works fine but the port scanner does not show port 4000 nor 2000.
In turn, or for another reason, my EFB (usually expecting flarm records through an Air Connect device) fails to show any traffic. Any hint that helps me to progress is most welcome.
Peter

closed 2018-01-17 by cyoung